prusa3d/PrusaSlicer #7780: Opening a previous version of PrusaSlicer damages configuration

combs, 2022-01-19:
### Version
2.4.0 and 2.3.3 released versions

### Operating system type + version
Ubuntu 20.04 x64

### 3D printer brand / version + firmware version (if known)
N/A (Mk3S)

### Behavior

- Open PrusaSlicer 2.4.0, configure it
- Quit PrusaSlicer 2.4.0
- Open PrusaSlicer 2.3.3. It says "Configuration bundle was loaded, however some configuration values were not recognized." "Most likely the configuration was produced by a newer version of PrusaSlicer [...]"
- It then says "PrusaSlicer configuration is incompatible" and requires you to exit or reconfigure
- Press "Exit PrusaSlicer"
- Open PrusaSlicer 2.4.0. It says "You are opening PrusaSlicer version 2.4.0 [...] The active configuration was created by PrusaSlicer 2.3.3 [...]"
- The configuration file was changed by PrusaSlicer 2.3.3 despite it knowing that the configuration was from a newer, incompatible version

### Desired behavior

- PrusaSlicer should not write its configuration until after the "PrusaSlicer configuration is incompatible" check

- Each version of PrusaSlicer should store its configuration in a separate path

#### Project File (.3MF) where problem occurs
N/A

n8bot, 2022-01-27:
Hmmm. Can you use the configuration snapshot tool to roll back (in the newer version) and then go back to the old version and you should have your old config.

bubnikv, 2022-01-28:
The issue here is that the old slicer modifiers PrusaSlicer.ini by saving its version into the "version" key before the "reconfigure" process is started. Once the old slicer is closed and new slicer is started, it reads the "version" key and it considers the old data to be out of date, because it was saved by an old slicer, which is not quite correct.

We shall fix slicer to save the "version" key first after the "reconfigure" step is over.

bubnikv, 2022-01-28:
Fixed in the upcoming PrusaSlicer 2.4.1-beta1, however the fix will start to work first after the upcoming PrusaSlicer 2.4.1-beta1 will become the older of the two PrusaSlicers one will switch between.

closed 2022-01-28 by bubnikv

combs, 2022-01-28:
Thanks so much!