prusa3d/PrusaSlicer #7781: Entering value in Thumbnail setting causes Prusa Slicer to crash

acorrow, 2022-01-20:
### Version
version_2.4.1-alpha0-24-ge0883910d

### Operating system type + version
Windows 11 Pro
Version 21H2
OS Build 22000.434

### 3D printer brand / version + firmware version (if known)
Prusa i3MK3S+ MMU2S

### Behavior
Adding the text 

16x16

To the Thumbnail field in the printer settings, causes Prusa slicer to crash.

Steps to repro:

1: Build and run
2: Edit printer (Cog)
3: Click in the thumbnail field
4: type 16x16

Expected; Able to set the size for the Thumbnail
Actual: Prusa Slicer crashes, every time. Without fail.

YuSanka, 2022-01-20:
You are right, thanks. We will fix it.

acorrow, 2022-01-20:
Confirmed fixed in version_2.4.1-alpha0-32-g6fed5c29a

Thanks for the quick response!

closed 2022-01-20 by acorrow