prusa3d/PrusaSlicer #7784: Slicer 2.4.0 retracting where 2.3.3 did not, ruins my first layer aesthetics.  Bolt knob top.

kevinmichaeldudley, 2022-01-20:
### Version
_Version of PrusaSlicer used goes here_
2.3.3 vs 2.4.0 

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ Mk3S+  current firmware.

### Behavior
* _Describe the problem_
* Using 2.3.3 the first layer, which is ALL perimeters, has NO retractions and prints a beautiful "knob" top when turn over.
* Using 2.4.0 the first layer has retracts and wipes which cause zits and artifacts to occur in the NOW not beautiful "knob" top.
* _Steps needed to reproduce the problem_
* Step thought the first layer using 2.3.3 and then step through the first layer using 2.4.0.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  * Slicer 2.3.3 beautiful smooth first layer.
[image]
Slicer 2.4.0 has retracts and wipes ruining the smooth first layer.
[image]

_Is this a new feature request?_
No it is a change between 2.3.3 and 2.4.0 which prevents me from using 2.4.0
I change settings and have been unable to replicate how 2.3.3 sliced the first layer.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

The named slicer files, 2.3.3 and 2.4.0 show the issue.
[Slicer 4 issue.zip](https://github.com/prusa3d/PrusaSlicer/files/7908067/Slicer.4.issue.zip)

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2022-01-21:
Seam positioning should be improved in the upcoming 2.4.1 release. It seems to improve the situation in your case:
[image]

bubnikv, 2022-01-28:
The seam placement issues will be improved with the upcoming PrusaSlicer 2.4.1-beta1 release.

closed 2022-01-28 by bubnikv