prusa3d/PrusaSlicer #7785: Platter view tool-tip text box blocked by mouse pointer.

ScrimpyD, 2022-01-21:
### Version
2.4

### Operating system type + version
Windows 11 Pro Insider Preview Build 22538.rs.prerelease.220114-1500
Windows 10 21H2

### Behavior
When in the Platter view the tool-tip text starts directly under the mouse pointer so you cannot see the first couple characters of whatever text is displayed. 

This does not happen with you are in the 3 Settings tabs. The tool-tip text shows up below the mouse pointer so you can read all the text.

_Is this a new feature request?_
I dont think so, more of an adjustment in the placement of the tool-tip text box in the Platter View.

lukasmatena, 2022-01-21:
Can you please post a screenshot so it is clear what exactly you mean? Thanks.

Jebtrix, 2022-01-22:
Are you using large or extra large mouse scheme in windows maybe?

ScrimpyD, 2022-04-07:
Sorry for the long delay.  I have my pointer size set to 3 from a scale that goes from 1 (very small) to 15 (large). I have a 4K monitor so that makes the default pointer size (1) very small. When I change to a size 2 the pointer is no longer covering the tool tip.
I was wondering if other users who had a slightly larger pointer size noticed this also.  Thank you for your time.

FidelCapo, 2022-05-03:
This issue should be improved in future versions.
Closing.

closed 2022-05-03 by FidelCapo