prusa3d/PrusaSlicer #7786: Not able to customize first color section when adding additional colored layers

mckinlk, 2022-01-21:
### Version
Version 2.4.0+win64

### Operating system type + version
OS Name	Microsoft Windows 10 Pro
Version	10.0.19043 Build 19043
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Type	x64-based PC

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Cannot change color of first color when adding additional layer colors; it is always dark green. Cannot right-click on first color to bring up the color select palette
[image]

I would expect to see the color select palette to customize the first color layers
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
N/A

neophyl, 2022-01-21:
The initial colour is taken from that defined in the filament profile.

You can either change the filament profile (non optimal) or you can swap to the plater view and on the right hand panel where the 3 dropdowns for the profiles in use are you will see a colour swatch next to the filament listed.  Click on that and you can change the filament colour.

With a search of issues you could also have found 
https://github.com/prusa3d/PrusaSlicer/issues/7403
or https://github.com/prusa3d/PrusaSlicer/issues/2346

mckinlk, 2022-01-21:
Seem like inconsistent UI to have to change first filament color in a completely different context. I agree on #7403 comfy option. Sorry for the duplicate.

closed 2022-01-21 by mckinlk