prusa3d/PrusaSlicer #7788: Printing with layer thicknesses greater than nozzle diameter

kontakt, 2022-01-21:
### Version
2.4.0

### Operating system type + version
Windows

### Behavior
Layer Height cannot be larger than nozzle diameter.
I get an error when trying to set the layer height greater than nozzle diameter. This is normally expected and good behavior, but I am printing with foaming PLA that expands after leaving the nozzle. This theoretically allows me to print with layer thicknesses in excess of nozzle diameter.
It would be helpful to have the option to override safety features.

Area5142, 2022-01-22:
I see the same problem with colorFabb varioShore TPU, where the foam expansion can be more than 120% and a height of more than the nozzle size would make sense.
As long as the print width is greater than the print height this should work fine for some models.

Maybe the error could be changed to a warning that states that this setting is only recommended for experts and the print may fail.

tyeth, 2022-01-22:
Looking for 250% nozzle size here. (ESUN PLA-LW)

closed 2026-09-29 by prusaslicer-bot[bot]