prusa3d/PrusaSlicer #7790: gcode viewer - error with CR LF line break types

SteWhite, 2022-01-21:
### Version
2.4.0
### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
mk3s 

### Behavior
If I download a gcode from Octoprint with dashboard plugin I've this issue:
"Configuration block closing tag "; prusaslicer_config = end" not found when reading xxxx.gcode"

With notepad++ I can see the difference:
the dashboard plugin add CRLF at the end on the line.
If I replacee CRLF with LF the error disappares.

#### GCODE file where problem occurs:
[octoprint_dash.zip](https://github.com/prusa3d/PrusaSlicer/files/7916637/octoprint_dash.zip)

[image]

Jebtrix, 2022-01-22:
This is an issue for the Octoprint plugin.

SteWhite, 2022-01-23:
> This is an issue for the Octoprint plugin to fix: https://github.com/j7126/OctoPrint-Dashboard/issues

Thanks for the reply but all other software opens that file except prusaslicer. I think it is enough to make a change and consider CRLF as LF.

Jebtrix, 2022-01-23:
I would still create an issue on the plugin, mixing EOL is really bad behavior regardless.

closed 2026-09-29 by prusaslicer-bot[bot]