prusa3d/PrusaSlicer #7791: [Bug] Reload from Disk "Unable to Reload" errors for AMF and 3MF files

Jebtrix, 2022-01-21:
### Version
Latest git build of master @ Revision: 959804c80ce449b7ea008494f32a75a287204742

### Operating system type + version
Win7/10 64bit

### Behavior
* Using AMF or 3MF mesh source files fail a Reload from Disk after saving and re-opening Project

* _Steps needed to reproduce the problem_
  * New Project
  * Import AMF or 3MF file/mesh
  * Save Project
  * Re-open Project
  * Try to Reload from Disk

Jebtrix, 2022-02-02:
Works fine, thanks @enricoturri1966

Jebtrix, 2022-02-02:
Looking at issue #7589 I played with more files. There is still an issue that some of my files have. Looking deeper it comes down to:

AMF files that don't include `<metadata type="name">...</metadata>` for object and volume elements Reload from Disk fails. Hope this helps.

[Issue_7991_Examples.zip](https://github.com/prusa3d/PrusaSlicer/files/7990164/Issue_7991_Examples.zip)

FrHePo, 2022-02-02:
If metadata for the amf-(or other)file are missing or erroneous why not open a user dialogue to allow the user to enter the correct data?

Jebtrix, 2022-02-03:
I missed today's commits, fixed with latest git master 086662b6a1c7d7a14e35a90e6d46445f6871ac24

closed 2022-02-03 by Jebtrix

FrHePo, 2023-11-28:
If metadata for the amf-(or other)file are missing or erroneous why not open a user dialogue to allow the user to enter the correct data?

> Works fine, thanks @enricoturri1966

No. it still doesn't work here.
Before I generate a new AMF-version with my cad-software reload from disk does not cause an error but after I generate a new AMF-version I get the usual error.
[image]