prusa3d/PrusaSlicer #7794: Variable layer height doesn't work when using modifier object.

ImTheBadWolf, 2022-01-22:
### Version
2.4.0

### Operating system type + version
windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Ender-3

### Behavior
Variable layer height doesn't work when using modifier object.
When the model contains modifier object, variable layer height doesn't work. 
After deleting the modifier, variable layer height works.

1.  Add modifier to model
2.  Add variable layer height to model
3.  Slice and check result => no variable layer height in the layers preview
4.  Remove modifier from the model and slice again
5. Variable layer height appears in the layers preview

[image]
[image]
[image]

#### Project File (.3MF) where problem occurs
.3MF files are not supported in issues
Here is zip file
[variable layer height issue.zip](https://github.com/prusa3d/PrusaSlicer/files/7918443/variable.layer.height.issue.zip)

Jebtrix, 2022-01-22:
Worked fine for me. Did you select Height for View mode?

[image]

ImTheBadWolf, 2022-01-23:
I tried selecting the height mode but it still doesnt show up in the preview.
[image]
\
[image]

ImTheBadWolf, 2022-01-23:
Update: 
I tried downloading the 3mf file that I uploaded here and in that file the variable layer height works. But when I try the same file that I uploaded, there the layer height doesnt work.

Jebtrix, 2022-01-23:
Trying to narrow this down it is an odd one. I netfabb fixed (just in case) objects, exported the STL meshes, started fresh project and can reproduce it consistently. For whatever reason the body4.stl mesh is the cause. The same problem can be reproduced if you add body4.stl as a part instead. Then if you change it to a modifier, variable height will not apply.

[image]

ImTheBadWolf, 2022-01-24:
Yeah and now if you would have saved the whole prusaslicer project as a new .3mf file and reopen that save, then the layer height will work even with that modifier.

closed 2026-09-29 by prusaslicer-bot[bot]