prusa3d/PrusaSlicer #7796: Invalid data

VjyedSC0UFMc8z, 2022-01-22:
### Version
Version 2.3.1+x64

### Operating system type + version
macOS Mojave v 10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s+

### Behavior
* _Describe the problem_
  * Vase mode is too strict when locking out other settings.
  * Vase mode interprets models with modifier meshes as multi-material models
* _Steps needed to reproduce the problem_
  * Import multi-shell STL (from f360 in this case)
  * Split into parts
  * Change text shell into modifier mesh
  * apply spiral vase setting
  * use modifier mesh to set infill settings for Aesthetics
* _Expected Results_
  * Cool pattern made inside text for legibility
* _Actual Results_
  * Prusaslicer says vase mode is only for single material objects
  * This is a single material, but still multi-part

[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs

[twist box vase.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7919082/twist.box.vase.3mf.zip)

Visit the URL to hire me. ;)
Thanks,
-Chris
:robot:

VjyedSC0UFMc8z, 2022-01-22:
My workaround was splicing two Gcode files together, but that takes too long.

Jebtrix, 2022-01-22:
You can do a Cut trick for your particular case. Vase is picky, but it has to be.

[twist box vase__cut_trick.zip](https://github.com/prusa3d/PrusaSlicer/files/7919435/twist.box.vase__cut_trick.zip)

Jebtrix, 2022-01-23:
The guard rails in Prusaslicer tend to leave out valid use case conditions. Any part or object not above total bottom layers height where vasing begins should be exempt. 

In your particular case  I would just change modifier type to Negative. Make the text mesh height = your 1st layer of 0.2mm. The text infill will become bridging fill. You can control bridge angle, flow, and speed to alter its look and tweak legibility. Its a compromise over what you wanted but it gets the job done.

[image]

VjyedSC0UFMc8z, 2022-01-26:
> change modifier type to Negative.

That's a good workaround too. Thanks for the input.

closed 2026-09-29 by prusaslicer-bot[bot]