prusa3d/PrusaSlicer #7797: PrusaSlicer Quits After Returning from Windows Sleep Mode

jywilson2, 2022-01-22:
### Version
 2.4.0+win64

### Operating system type + version
Windows 10 (build 19043), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3s

### Behavior
If PrusaSlicer is left open in the background after a period of time it will shut itself down.  The same problem occurs when returning from Windows sleep mode.

#### Project File (.3MF) where problem occurs
Not applicable.  This problem occurs even when no models are loaded.

Jebtrix, 2022-01-22:
Assuming you installed Prusaslicer in default location launch this from a command prompt:
`"%ProgramFiles%\Prusa3D\PrusaSlicer\prusa-slicer-console.exe" --loglevel=9 > "%HOMEDRIVE%%HOMEPATH%\Desktop\prusa_lvl9_log.txt" 2>&1`

prusa_lvl9_log.txt file will be generated on your Desktop and you can attach results for devs

jywilson2, 2022-01-23:
The requested log file is attached.

Thanks,

Jim W
[prusa_lvl9_log.txt](https://github.com/prusa3d/PrusaSlicer/files/7919714/prusa_lvl9_log.txt)
.

Jebtrix, 2022-01-23:
Definitely nothing interesting in that log file. Try running with software opengl instead of gpu acceleration:
`"%ProgramFiles%\Prusa3D\PrusaSlicer\prusa-slicer-console.exe" --sw-renderer --loglevel=9 > "%HOMEDRIVE%%HOMEPATH%\Desktop\prusa_lvl9_log.txt" 2>&1`

jywilson2, 2022-01-23:
I took a quick look and I don't think this log will of any help either.  

I am more familiar with using gdb on Linux to capture call stacks of crashed executables.  Is there way to do something similar in Windows, perhaps with VS Code?

[prusa_lvl9_log.txt](https://github.com/prusa3d/PrusaSlicer/files/7919928/prusa_lvl9_log.txt)
?

Jebtrix, 2022-01-23:
Well the real test was the software renderer. If it still crashes that narrows the causes of GPU/driver. You can try the latest WinDbg Preview and attach to process: [ https://www.microsoft.com/en-us/p/windbg/9pgjgd53tn86](https://www.microsoft.com/en-us/p/windbg/9pgjgd53tn86)

n8bot, 2022-01-23:
Are you using the dark mode setting in PrusaSlicer?

jywilson2, 2022-01-24:
Yes, I am using Dark Mode. I will try disabling it.

If that doesn't help, I will use WinDbg as suggested to get a stack trace.

n8bot, 2022-01-24:
Try updating windows to the 21h2 update.

jywilson2, 2022-01-24:
That did it! Updating Windows to 21h2 has allowed me to run in Dark mode
with GPU support.

One thought would be to add a check for the version of Windows in the
PrusaSlicer install program, or at least provide a warning of possible
issues at startup of PrusaSlicer.

I will close the issue.

Thanks n8bot!

Jim W.

closed 2022-01-24 by jywilson2

n8bot, 2022-01-24:
Oh nice! Haha. I removed my comment because I thought it might be a red herring. Looks like my suspicions were correct. Microsoft did some things in November that really messed with some of the work Prusa Team did leading up to release.