prusa3d/PrusaSlicer #7799: Switching tabs with Ctrl+Shift+Tab collapses sidebar

the-ress, 2022-01-23:
### Version
2.4.0+win64

### Operating system type + version
Windows 11 (21H2, build 22000.466)

### Behavior
1. Start on the Plater tab with the sidebar visible
2. Press Ctrl+Tab to switch to the Print settings tab
3. Press Ctrl+Shift+Tab

Expected result: Plater tab becomes visible again. Sidebar is still visible.
Actual result: Plater tab becomes visible again. Sidebar gets collapsed.

[image]

lukasmatena, 2022-01-24:
Thanks. Can you please try this build if the behaviour is better? Apparently the way of handling the keyboard modifiers might be improved. https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.4.1-alpha0+19-lm-keymodifiers-win64-g3c619bf8d-202201171528.zip

the-ress, 2022-02-05:
It's better; I couldn't reproduce the bug in that build. Thanks.

the-ress, 2023-12-03:
I noticed it's back in 2.7.0.

FidelCapo, 2024-01-17:
This fix was not part of version 2.7.0. It will be part of the next version.
Closing.