prusa3d/PrusaSlicer #7800: Feature request: More recycle codes in the shape library

VorpalBlade, 2022-01-23:
### Version
Version 2.4.0+linux-x64-GTK3

### Operating system type + version
Arch Linux (rolling release)

### 3D printer brand / version + firmware version (if known)
Prusa MK3s (but irrelevant)

### Behavior
The addition of the shape library, and specifically that of recycling symbols is awesome. But so far only PETG and Other are available. While I know the European system is really inadequate for the types of plastics used for 3D printing, it would be useful to have more specific symbols, either by adopting the [Chinese system](https://en.wikipedia.org/wiki/Recycling_codes#List_of_Chinese_codes_for_plastics_products) or by adding an extra second information line as to which "other" filament it is.

This can also be useful for those who recycle themselves with a home made filament extruder for example. (Still a rather niche use case, but interest does seem to be growing).

Personally I would like to see at least symbols for PLA (extremely common in 3D printing) and PA (something I use myself).

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
N/A

rtyr, 2022-01-24:
Are you aware that you can add your own models to the gallery?

foreachthing, 2022-01-24:
This should get you started: https://github.com/foreachthing/RecyclingSymbols

lukasmatena, 2022-04-13:
We added a symbol for PLA in [PrusaSlicer 2.4.2-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc2). We now have symbols for PLA and PETG which are the most used materials, anyone can add whatever else they want. We do not want to add everything imaginable to not clutter the gallery too much. Can I close the issue?

VorpalBlade, 2022-04-13:
@lukasmatena Fine by me. It covers what I use. And as I understand it (and I'm not at all an expert) when it comes to some plastics such as nylons and TPUs there are many variants which would make it even more complicated.