prusa3d/PrusaSlicer #7801: AppImage missing icon

VorpalBlade, 2022-01-23:
### Version
Version 2.4.0+linux-x64-GTK3 (officially AppImage)

### Operating system type + version
* Arch Linux (rolling release)
* Desktop environment: Cinnamon 5.2.7-2

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* Unlike other AppImages (e.g. FreeCAD) PrusaSlicer doesn't seem to have an icon in the taskbar when running:

[image]

As can be seen, PrusaSlicer gets a blank icon, while FreeCAD gets an icon just fine. Both are running from AppImages.

I looked into the AppImage and it seems that both have correct .desktop files and .png in the root. The only possibly relevant difference I can see is that FreeCAD has it's icon also in usr/share/icons, while PrusaSlicer is missing that. According to https://docs.appimage.org/reference/appdir.html#usr-subdirectory "Contains at least one, but often a set of icons for the main application. The icons are referred to by the root desktop file". This missing is probably the issue, though I haven't tried to modify the AppImage to test this. If that is the case, it should be easy to fix this.

_Is this a new feature request?_
No, it is a weird bug of some sort.

#### Project File (.3MF) where problem occurs
N/A. Happens without any project loaded.

anngedeus, 2022-01-23:
Hello I would be happy to fix this issue

anngedeus, 2022-01-24:
I have a few questions how did you go about looking in the AppImage?

Area5142, 2022-01-25:
The files in an AppImage can be extracted simply by adding a command line option _--appimage-extract_ to the AppImage:

`./PrusaSlicer-2.4.0+linux-x64-202112211614.AppImage  --appimage-extract`

This will place the extracted files in directory _./squashfs-root_.

To pack/reassemble the AppImage again, use the _[appimagetool-x86_64.AppImage](https://appimage.github.io/appimagetool/)_ tool:

`./appimagetool-x86_64.AppImage ./squashfs-root`

Result is the AppImage: ./PrusaSlicer-x86_64.AppImage

This is how I make an AppImage after compilation on Linux.

bubnikv, 2022-01-28:
There should be a "Desktop integration" menu item under Configuration, which shall install the particular desktop and icon files. Did it help?

bubnikv, 2022-01-28:
@VorpalBlade 

> Using that method of desktop integration seems to work better with respect to getting correct icons

Better or is it resolved?

Regarding the AppImage integration daemon, you are the first person I am aware of actually using it. It is not being distributed by the common Linux distros.

@probonopd Do you have any idea why our AppImage is not being integrated by the daemon?

probonopd, 2022-01-28:
>> Normally I use https://docs.appimage.org/user-guide/run-appimages.html#appimagelauncher for desktop integration. With that method PrusaSlicer doesn't get any icons. Other AppImages get icons just fine.

> @probonopd Do you have any idea why our AppImage is not being integrated by the daemon?

@VorpalBlade, @bubnikv that would be a question for @TheAssassin at https://github.com/TheAssassin/AppImageLauncher/.

TheAssassin, 2022-01-28:
https://github.com/AppImage/libappimage/issues/165

probonopd, 2022-01-28:
Let's see:

```
# Get and extract PrusaSlicer AppImage
% wget "https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.4.0/PrusaSlicer-2.4.0+linux-x64-202112211614.AppImage"
% chmod +x PrusaSlicer-2.4.0+linux-x64-202112211614.AppImage
% ./PrusaSlicer-2.4.0+linux-x64-202112211614.AppImage --appimage-extract

# Check which icon(s) it specifies
% grep -r Icon squashfs-root/*.desktop
squashfs-root/PrusaSlicer.desktop:Icon=PrusaSlicer
squashfs-root/qt-dab.desktop:Icon=qt-dab

# Check where those icons are located in the AppImage
% find ./squashfs-root -name 'PrusaSlicer.png' -or -name 'qt-dab.png'
./squashfs-root/PrusaSlicer.png
./squashfs-root/usr/share/icons/hicolor/256x256/apps/qt-dab.png
./squashfs-root/usr/bin/resources/icons/PrusaSlicer.png
./squashfs-root/qt-dab.png
```

So it seems like the `PrusaSlicer.png` icon is missing from `usr/share/icons/hicolor/256x256/apps/` in the AppImage, which is the location from which AppImageLauncher is trying to pick it up.

Adding `usr/share/icons/hicolor/256x256/apps/PrusaSlicer.png` in the AppImage should fix it.

TheAssassin, 2022-01-28:
Workaround, though. Yes, will work, but an already existing `.DirIcon` should just be deployed as well. libappimage already knows how to read an icon's size, it shouldn't be too hard to find the right target directory.

VorpalBlade, 2022-01-28:
> Regarding the AppImage integration daemon, you are the first person I am aware of actually using it. It is not being distributed by the common Linux distros.

It is available in AUR for Arch Linux. I tried it and it works quite well, with the exception of PrusaSlicer. What is extra nice is the built in support for updating with partial download using zsync (for AppImages that support it, such as FreeCAD, PrusaSlicer doesn't have the required metadata).

> Better or is it resolved?

One thing that doesn't seem to work with the built in integration is the file association for 3mf. With AppImageLauncher I believe it worked. AppImageLauncher at least works properly with FcStd files for FreeCAD. With the PrusaSlicer built in integration I get errors about that the file that 3mf is associated with is no longer available. Might be due to some left overs from the previous integration? Not sure.

> Workaround, though. Yes, will work, but an already existing `.DirIcon` should just be deployed as well. libappimage already knows how to read an icon's size, it shouldn't be too hard to find the right target directory. No spec mandates the existence of files in `usr/share/icons`, only files in the root dir are required.

Not sure on this one, but reading https://docs.appimage.org/reference/appdir.html#usr-subdirectory (subsection share/icons) there is some vague wording that seems to indicate that there should be an icon in usr/share/icons and that "Icon themes placed in this directory are copied to the system during so-called desktop integration.".

Does it hurt to add such an icon? Since the AppImage uses squashfs I can't imagine it will really take up much more space. You could presumably even hardlink (or symlink) the png into multiple locations for almost no overhead.

TheAssassin, 2022-01-28:
> Does it hurt to add such an icon? Since the AppImage uses squashfs I can't imagine it will really take up much more space. You could presumably even hardlink (or symlink) the png into multiple locations for almost no overhead.

No, of course it doesn't. Even if it was a duplicate, who cares about a few kB of storage? There should be a fallback nevertheless.

brjhaverkamp, 2024-10-09:
Hello all, I am using Prusaslicer 2.8.1, and I see the icon is still not there in the taskbar/dash. I'm on Ubuntu 24.04, using appimagelauncher to integrate the appimage in the system.
Is there any blocker for fixing this issue?

closed 2026-09-29 by prusaslicer-bot[bot]