prusa3d/PrusaSlicer #7802: [fature request] Insert custom Gcode befor and after perimeters

MichaelOEstreicher, 2022-01-23:
I would like a option to insert custom gcode before and after  perimeters.
My printer supports ringing cancellation. I wold like to switch it off to speed up the
infill printing, and switch it on during the perimeters for better printing results.

fe60, 2022-01-24:
That's a pretty specific use case. I doubt this is relevant for many users.

Proposal to fulfill your requirement with existing options:
You can insert the machine codes for switching on/off the ringing cancellation into the generated gcode-file.
Prusa Slicer marks the section where infill is printed with a comment.
<details>
<summary>Screenshot showing beginning of infill</summary>

[image]
</details>

Printing infill ends when the layer is finished.
<details>
<summary>Screenshot showing end of layer</summary>

[image]
</details>

Using an editor offering regular expressions for search and replace, you can add the missing commands with the following strings:
Search: `(;TYPE:Internal infill.*?;LAYER_CHANGE)`
Replace with: `RINGING_OFF\n\1\nRINGING_ON`
This works for me in Notepad++.
<details>
<summary>Screenshot showing search dialogue in Notepad++</summary>

[image]
</details>
Some comments about the search and replace strings:

- `(...)` brackets mark a capturing group, referenced in replace string with `\1` to insert the match [1]
- `;TYPE:Internal infill` comment in front off gcode for infill
- `.*` look for an arbitrary amount of arbitrary characters [2,3]
- `?` avoids matching everything from the first infill to the last layer change [4]
- `;LAYER_CHANGE` comment before next layer => after infill
- `\n` newline to be inserted after first and before second command
- `RINGING_OFF|ON` machine codes (to be replaced by the real ones) to switch off/on ringing cancellation

https://en.wikipedia.org/wiki/Regular_expression#Examples
[1] 2nd row explains `( )`
[2] 1st row explains `.`
[3] 6th row explains `*`
[4] 5th row (second occurence of `?`) explains `?`

MichaelOEstreicher, 2022-01-24:
Thank you very much for the hint to a working solution.

My system is running Ubuntu.
For this reason I now use a executable post processing script with a sed call.
with the following content:

`#!/bin/sh
sed -i -e s/\;TYPE\:External\ perimeter/M593\ F8.53\;\ ringing\ cancelation\ 8.53Hz/\
       -e s/\;TYPE\:Perimeter/M593\ F8.53\;\ ringing\ cancelation\ 8.53Hz/\
       -e s/\;TYPE\:Overhang\ perimeter/M593\ F8.53\;\ ringing\ cancelation\ 8.53Hz/\
       -e s/\;TYPE\:Solid\ infill/M593\ F-1\;\ ringing\ cancelation\ off/\
       -e s/\;TYPE\:Internal\ infill/M593\ F-1\;\ ringing\ cancelation\ off/\
       $1`
This script is executed automatically after export. 
It replaces every occurrence ";TYPE:External perimeter", ";TYPE:Perimeter", and ";TYPE:Overhang perimeter" with the code for the ringing cancellation activation "M593 F8.53"
and every occurrence of ";TYPE:Solid infill" and ";TYPE:Internal infill" with the code for the ringing cancellation deactivation "M593 F-1"

[image]

closed 2022-01-24 by MichaelOEstreicher

foreachthing, 2022-01-25:
@MichaelOEstreicher instead of replace the comments (";TYPE:External perimeter"), you could/should add it below (or before). That way the GCode Viewer still knows what to do - if you use that at all ;-)

MichaelOEstreicher, 2022-01-25:
Thank you again.
The modified sed call no looks like this.
`sed -i -e /erimeter*$/aM593\ F8.53\;\ ringing\ cancellation\ 8.53Hz  -e /infill*$/aM593\ F-1\;\ ringing\ cancellation\ off  $1`
It inserts the gcode for the ringing cancellation after every line which ends with "erimeter"
and the gcode for switching of the ringing cancellation after every line which ends with "infil"

bubnikv, 2022-01-28:
You can enter the regular expression substitution directly into the upcoming PrusaSlicer 2.4.1-beta1 to be released soon.

bubnikv, 2022-01-28:
[image]

carlosjln, 2024-01-25:
Just what I was looking for 😩 it only occurred to me now that it would be helpful to turn off linear advanced feature for the internal infill.

I came up with this regex to match the internal infill block `(;TYPE:Internal infill\n)(.*?)(;TYPE:.*?\n|$)` and replace with `${1}M900 K0\n${2}M900 K0.6\n${3}`

K0.6 is just the linear advance value I'm using, but I'm posting it here so that other people can get the idea.