prusa3d/PrusaSlicer #7808: Import FreeCAD's FCStd format

eduncan911, 2022-01-24:
### Version
Version 2.4.0+linux-x64-GTK3

### Operating system type + version
Pop_OS 21.10 (Ubuntu 21.10) + Regolith

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Cannot Import -> FCStd format

CURA, Simplify3D, etc all support importing an FCStd file.  This is a feature request to add the same support here.

FreeCAD and OpenSCAD are the two most dominant CAD packages on Linux, because of the limited-to-no support of the expensive CAD packages.  

Not to mention they are both free and open source, just like Prusa's moto is!

_Is this a new feature request?_

YES

#### Project File (.3MF) where problem occurs
N/A

n8bot, 2022-01-24:
Is FCStd a step-like file? As in, solid data not mesh?

Either way, it sounds like an interesting project to tackle.

eduncan911, 2022-01-24:
CURA has FreeCAD, SolidWorks, etc integration.  It looks like these plugins for CURA all use the "installed software" to convert/model with.  Here's a nice workflow shown for SolidWorks, with a single GUI popup:

https://www.youtube.com/watch?v=Ixm2crbtCtg

Basically, it looks to import a single Part only, which makes sense as who knows what part you want to print if you open an entire assembly.

An `FCStd` file can be used as an Assembly, or individual Parts - just like SW.

So far, it looks like it uses the external program itself to open, and then export, an STL mesh.  That's pretty simple actually.

---

TL;DR

If you need help on the command line format/options for FreeCAD to open, convert, and save/output an STL, let me know and I'll own that part.

HaasNL2, 2022-01-25:
Would be awesome if Prusaslicer could save the slicer settings to the Freecad file, this way when we update the model we just have to to check and hit print.
No more transfer files like 3fm needed.
Being able to select the bodies you want print is cool too, but Prusa slicer already has a nice way in disabling the bodies that you don't want to print.

eduncan911, 2022-01-25:
> Being able to select the bodies you want print is cool too, but Prusa slicer already has a nice way in disabling the bodies that you don't want to print.

True, but have you seen some of these 3D printer step files?  with over 100 bodies?  And that's if they happen to name tings properly.  lol

bubnikv, 2022-01-25:
AFAIK Cura only supports importing CAD models using a commercial CAD kernel, for which you have to pay.

n8bot, 2022-01-25:
I doubt there are any non-free licensing requirements for the FreeCad format, tho.

Also, IIRC, there is no license needed to decode STEP files, but I might be wrong about that.

eduncan911, 2022-01-25:
There is no license or commercial software needed.  You run the CAD binaries via exec to convert objects.

> AFAIK Cura only supports importing CAD models using a commercial CAD kernel, for which you have to pay.

For SolidWorks files, yes as there is no free version.  

Fusion360 I am not sure if they have a free CLI conversation program.

But again, this issue is for importing FreeCAD FCStd conversions - and there is no commercial engine or license or whatever.

The CURA plugin that does FreeCAD imports points to the installed `/bin/FreeCADcmd` binary for CLI conversions of Part Bodies to STL.  You can even configure the mesh resolution (though not sure if it's through defaults only or also through CLI).

eduncan911, 2022-01-25:
I know nothing about Slicer/PrusaSlicer/SuperSlicer internals, and mostly likely have no spare time to dig in.

However, I'm more than happy to research and provide all the CLI binary vars and envs needed.

Also, I can test and verify the FreeCAD imports via Linux, macOS, and Windows default installs.

HaasNL2, 2022-01-25:
> > Being able to select the bodies you want print is cool too, but Prusa slicer already has a nice way in disabling the bodies that you don't want to print.
> 
> True, but have you seen some of these 3D printer step files? with over 100 bodies? And that's if they happen to name things properly. lol

Reverse-select and hide would fix that pretty easy. Although I there is no reverse select atm. So having ether a pop-up or reverse select would probably be a solution to this. 
@eduncan911 Would be best of it's not an Import function but an native support function, so safe your settings in the Freecad file. Which would not be that difficult since as far as I think it's all XML layout, Prusaslicer settings / 3fm and fcstd settings. 
https://wiki.freecadweb.org/File_Format_FCStd 
https://3mf.io/specification/
[image]

eduncan911, 2022-01-25:
@HaasNL2 i don't see a need to "save PrusaSlicer settings in FreeCAD FCStd" to go back to FreeCAD.  That's what 3mf is for. 
 Besides, if you do that and you already had FreeCAD open, any Saves would just overwrite the file.  That's a bad workflow IMO, as it can easily get overwritten by FreeCAD saves.

Let's take this one step at a time... Importing FCStd first.  Then maybe a Reverse Select as you mentioned...

bubnikv, 2022-01-28:
> There is no license or commercial software needed. You run the CAD binaries via exec to convert objects.

I see, to implement we would "just" need to buy all various commercial CAD systems, integrate, test and maintain. That sounds like a challenge. 

CURA has the benefit of 3rd party plugins, where these plugins are contributed and neither tested nor maintained by Ultimaker. We don't have such a plugin framework.

> The CURA plugin that does FreeCAD imports points to the installed /bin/FreeCADcmd binary for CLI conversions of Part Bodies to STL. You can even configure the mesh resolution (though not sure if it's through defaults only or also through CLI).

We may consider to implement such a functionality if there is enough demand.

> CURA, Simplify3D, etc all support importing an FCStd file.

I would be surprised if S3D imported FCStd file. If so, then by calling the FreeCADcmd binary as you suggested.

luzpaz, 2023-08-05:
Any traction on this ?

sslupsky, 2024-10-03:
I would like to suggest the next major version of PrusaSlicer be released as a workbench for FreeCAD.  FreeCAD is cross platform and open source and the mutual benefits to the Prusa and FreeCAD communities would be greater than the sum.

luzpaz, 2024-10-19:
Anybody interested in spearheading it ?

sslupsky, 2026-09-02:
IMO, this feature request is still desired and it does not appear to be mentioned in the 3.0.0alpha11 release notes.  Is there some sort of business strategy that explains why Prusa does not support FreeCAD more?  FreeCAD is getting better and better and very usable.

Jan-Soustruznik, 2026-10-19:
Hello, thank you for the update. Following our new [policy,](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I am closing this feature request and converting it into a Discussion thread. This will allow us to hear what the community thinks about this feature and provide a place for others to upvote it. Have a great day.