prusa3d/PrusaSlicer #7810: Start up error

ColinWise, 2022-01-24:
### Version 2.4.0 and 2.4.0 rc
_Version of PrusaSlicer used goes here_

Im sure this is something simple ,when i o[pen prusa i get this problem .(screenshot uploaded) i click ok and everything appears normal. 
i'm not sure this is the correct place to report this or ask for help so i appologise in advance.
### Behavior

* When i start the prusa software i get an error , 
*
 

colinwise11@gmail.com

[computer info.zip](https://github.com/prusa3d/PrusaSlicer/files/7928969/computer.info.zip)

bubnikv, 2022-01-25:
I am afraid you happened to run PrusaSlicer before under a different user name, thus the file reported seems to be not accessible by the user under which you executed PrusaSlicer the last time.

You may want to delete the whole AppData\Roaming\PrusaSlicer directory and re-run the installation wizard.

ColinWise, 2022-01-25:
Thanks for the quick  response.. and thanks for such amazing software.
Best regards
Colin

closed 2024-02-19 by lukasmatena