prusa3d/PrusaSlicer #7811: Nozzle temperature

wschwack, 2022-01-25:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

wschwack, 2022-01-25:
When I selected Prusament ASA filament in the slicer, exported the GCODE, uploaded it in octoprint and started the print, the nozzle temperature was set to 170 °C instead 260 °C. It was repeatable 5 times.

wschwack, 2022-01-25:
> When I selected Prusament ASA filament in the slicer, exported the GCODE, uploaded it in octoprint and started the print, the nozzle temperature was set to 170 °C instead 260 °C. It was repeatable 5 times.

Now I realized, the same was with PLA. Copying the GCODE file onto the stick and starting from stick (MINI+), the nozzle temperature generally was set to 170 °C.

neophyl, 2022-01-25:
It is designed to do this.  Please examine the contents of the printer setting Start gcode.  Inside you will find a 2 step warmup.  The nozzle is first heated to 170 so it can reach thermal equilibrium for mesh levelling.  It is 170 as generally at that temperature filament wont leak out.  Once the levelling is complete the printer will warm up to your set configured filament temperature.  
If mesh levelling was done at full temp then you get little spots of filament at the mesh points messing up your print surface.

wschwack, 2022-01-25:
> It is designed to do this. Please examine the contents of the printer setting Start gcode. Inside you will find a 2 step warmup. The nozzle is first heated to 170 so it can reach thermal equilibrium [...]

Now I understand. Obviously, that's new with the MINI.

rtyr, 2022-01-25:
Answered. Intended behavior. Closing.

closed 2022-01-25 by rtyr