prusa3d/PrusaSlicer #7812: Show center of gravity after slicing

paloss, 2022-01-25:
### Version
2.5

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
mk3s - latest

### Wanted behavior
Show the center of gravity after slicing.

I'm making a lot of statues. For now i open the stl file in meshmixer, check the center of gravity and then use modifiers in prusa slicer to add a higher density infill on 1 side / some places to try and align the center of gravity so it becomes as stable as possible. 

Its a bit of a gamble how much extra infill weight i need to put in on 1 side, so sometimes its great and sometimes its more trial and error.

For me it would be very helpfull if i can see the center of gravity after slicing so i know if i'm on the right track instead of just hitting print and gamble.

_Is this a new feature request?_
YES

foreachthing, 2022-01-25:
You could do it like this:
1. Export GCode
2. Open GCode in Viewer (PrusaSlicer -> File -> GCode Preview...
3. In Viewer: File -> Export Toolpath as OBJ ...
4. Import OBJ in [Blender](https://blender.org)
5. Select only imported Model and: 
[image]
6. Move to model in X and Y to zero.
7. Confirm it won't tip over 😼

n8bot, 2022-01-25:
I did not realize how amazing that export to obj function is. I tried it for the first time right now.

It exports EXACTLY what you see in the preview -- colours and all. This is incredible. While the extrusions are not physically perfectly accurate... I wonder if we could do FEA on this! LOLOL. Probably not at all useful. Remeshing it would be difficult with all the sharp thin edges. If we could export an accurately shaped model of the extrudate (which would be much too many polygons to use for the visualization)... we could use these exports for experiments with FEA.

foreachthing, 2022-01-25:
OT:
FEA??? Are you ...! What software do you have and what do you do with it?
Just imagine some larger model with gyroid infill ... elements count through the roof (like my test model with >8M faces)!! 🤣 Unless you have some awesome remesher - which I'd like to see! Tet-wild might help some, thinking about it.

n8bot, 2022-01-26:
> OT: FEA??? Are you ...! What software do you have and what do you do with it? Just imagine some larger model with gyroid infill ... elements count through the roof (like my test model with >8M [...]

LOL... I never thought about it too hard. Is it really that many more mesh elements than normal in an FEA study? It depends what the study is I guess. I'm thinking basic linear static stress. But, yeah, I bet you're right. The number of elements would be huge to get good results. I guess one could isolate a small part of the sliced object, and use frictionless constraints or whatever to simulate the rest of the body in the FEA.

But then again, this is a problem that many smarter minds than my own have been working on for at least a decade... so I doubt getting an OBJ of toolpaths is really the hard part of all this.

I digress from the topic at hand. It seems like the approximated toolpath shapes would be more than enough to get an accurate CofG.

bubnikv, 2022-01-28:
For my reference, implementation in upstream Slic3r.
https://github.com/slic3r/Slic3r/commit/f442fa77820bb87dc87b4545552af26b456d0c1b

rlhelinski, 2022-11-28:
I too would find a center of gravity (cog) feature very useful. I hope this gets merged in from upstream soon.

foreachthing, 2022-11-28:
2.6 alpha has this feature:

[image]

kubispe1, 2023-02-01:
Closing this issue as resolved.

closed 2023-02-01 by kubispe1