prusa3d/PrusaSlicer #7813: Don't draw wipe tower in 3D editor view when using only one extruder

martijnvankekem, 2022-01-25:
### Version
PrusaSlicer-2.4.0+win64-202112211614

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+ MMU2S
Firmware v3.10.1

### Behavior
When printing parts while using a single extruder (not in single mode, but in MMU mode with only 1 extruder selected), the wipe tower doesn't appear after slicing, however, it does appear in the 3D editor view before slicing.

This is unwanted since the wipe tower is included when clicking the 'Arrange' button for example, and will try to fit the wipe tower on the bed, even though it's not sliced. When using a model that covers the entire bed, that pushes those models off the bed.

[image]

The expected behavior would be to only show the wipe tower in the 3D editor view when the wipe tower will appear after slicing. In other words: only show the wipe tower in the 3D editor view when 2 or more extruders are used for that print. If not, just hide it.

n8bot, 2022-01-25:
You could probably disable the wipe tower option in the print settings tab. That should disable the wipe tower manually for cases such as this.

martijnvankekem, 2022-01-25:
> You could probably disable the wipe tower option in the print settings tab. That should disable the wipe tower manually for cases such as this.

@n8bot that's true, and that's what I do at the moment. However, I use the single extruder for literally all of my single-color prints, since I personally think the single mode of the MMU is a bit of a hassle. It requires me to select an extruder, and I'm mostly printing via Octoprint remotely.

It's a nice temp fix, but it would be great (and also kind of expected behavior) to disable the wipe tower when it isn't necesarry.

bubnikv, 2022-01-28:
> Additionally: when going back to the 3D editor view after slicing, the wipe tower isn't there anymore.

Before slicing the wipe tower size is not known. After slicing it is known to be not used, thus it disappears.

> However, when clicking 'Arrange' again, the wipe tower appears again, only OUTSIDE of the print bed after arranging.

Once you click Arrange, the objects are reshuffled and the wipe tower size is switched to a "don't know" state.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena