prusa3d/PrusaSlicer #7814: Firmware Crash After Removing SD Card

dinosaurus-rex, 2022-01-25:
### Version
2.4.0+win64

### Operating system type + version
Windows 11 Insider Preview 10.0.22538.1010 (rs_prerelease)

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S+
- Upgraded from Original Prusa i3 MK3S
- I use both the smooth PEI and satin steel sheets
MMU2S
- Upgraded the PTFE holder at the back of the MMU2S to this: https://www.prusaprinters.org/prints/111195-prusa-mmu2s-rear-ptfe-holder-m10
- Upgraded to this filament buffer: https://www.prusaprinters.org/prints/77054-mmu2s-buffer-inspired-by-rmu-buffer

### Behavior
- Completed a print using the "Original Prusa i3 MK3S & MK3S+ MMU2S Single 0.6 nozzle" profile
- Removed the SD card with the machine still on
- 2 or 3 seconds later the printer beeped once and the LCD said "FW crash detected" with no further details
- Pressed the knob and the message went away and the printer seems to be completely back to normal
- I don't know if this is reproduceable because it has only happened once after 37d 07h 48m of printing

#### Project File (.3MF) where problem occurs
[spacer tight.zip](https://github.com/prusa3d/PrusaSlicer/files/7936508/spacer.tight.zip)

neophyl, 2022-01-25:
Think this should be in the MK3 github and not the Slicer one.  https://github.com/prusa3d/Prusa-Firmware/issues

dinosaurus-rex, 2022-01-25:
Oops. Thanks, I'll close this and put it there.

dinosaurus-rex, 2022-01-25:
Actually I think it is already posted there haha

closed 2022-01-25 by dinosaurus-rex