prusa3d/PrusaSlicer #7815: Q: Is it possible to generate a g-code that is not heating up the nozzle on bridging?

Kreyren, 2022-01-26:
I use Creality ender-3 set on 250mm/s printing and nozzle 260C (the extruder starts skipping, bcs the hot end is unable to melt the plastic fast enough if the temperature is set lower) where for bridging i have to slow down to 10mm/s to avoid print failures which is overheating of the plastic for wider areas -> Is it possible to generate a g-code that won't allow the printer to heat up the hot end when it is doing bridging?

### Version
_Version of PrusaSlicer used goes here_

Version 2.4.0

_Use `About->About PrusaSlicer` for release versions_

Version 2.4.0

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

ain't dev

### Operating system type + version
_What OS are you using, and state any version #s_

GNU Guix Nix/Linux (f92d758f1dfbcb18b426795ed53666d4aa70e0a1)

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

ain't doing 3D rendering issues

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Stock Creality Ender-3, Marlin Ver 1.0.1 (2020-04-25)

### Behavior
* _Describe the problem_

Above

* _Steps needed to reproduce the problem_

Get any 3D printer which steppers are faster then hot-end's ability to melt plastic in time to not cause extruder skipping and try to print anything with +- 5x2 cm bridge area.

  * _If this is a command-line slicing issue, include the options used_

ain't

* _Expected Results_

Do not heat up the hot end when bridging

* _Actual Results_

annoying popping noises of overheated PLA

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  
  
[image]

_Is this a new feature request?_

yes

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.4
[modular_planter-wall_side.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7938904/modular_planter-wall_side.3mf.zip)
2.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

DISCLAIMER: Use on your own risk, i am not responsible for you breaking your printer or yourself with these settings.

[modular_planter-wall_side.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7938931/modular_planter-wall_side.3mf.zip)

Undo the `.zip` append from the file name

fe60, 2022-01-26:
The issue tracker is intended to report bugs and to request features.

Let's assume it's a feature request:
I don't think it makes sense to implement an option to switch off the hotend while printing bridges. It takes a while to cool down the hotend and it will again take a while to heat it up, once the nozzle passed the bridging section. So there would be sections in the bridge where the filament is still too hot, and the first part of the "regular" printing will be done with filament not heated up sufficiently.

To tackle needs similar to yours, there is already a bridge specific fan speed [1].

[1] https://help.prusa3d.com/en/article/cooling_127569#fan-settings

Kreyren, 2022-01-26:
> I don't think it makes sense to implement an option to switch off the hotend while printing bridges. It takes a while to cool down the hotend and it will again take a while to heat it up, once the nozzle passed the bridging section. So there would be sections in the bridge where the filament is still too hot, and the first part of the "regular" printing will be done with filament not heated up sufficiently. -- @fe60 (https://github.com/prusa3d/PrusaSlicer/issues/7815#issuecomment-1022170775)

The filament is fed at 250mm/s prior to bridging that seems to cool down the filament below glass transaction temperature ("Tg") in 2.4 sec when the hot end is not heating up to satisfy the temperature for bridging at slower speed.

I do agree that being able to lower the hot end temperature on bridging would be more appropriate to avoid going below Tg, but i ain't convinced that the reported temperature is accurate or that the hot-end is unable to transfer the heat to the filament fast enough as going on 300mm/s with 260C causes slipping and the filament can't be pushed in manually.

So i assume this situation happening (Credit CNCKitchen):

[image]

> To tackle needs similar to yours, there is already a bridge specific fan speed [1]. -- @fe60 (https://github.com/prusa3d/PrusaSlicer/issues/7815#issuecomment-1022170775)

Insufficient, the fan speed is already at 100% at all times.

fe60, 2022-01-28:
I still don't think it makes sense to add a corresponding feature to Prusa Slicer.

For your special use case you could add the machine codes de-/activating the heating after generating the gcode.
Prusa Slicer adds comments to the gcode, stating what will be printed next.
`;TYPE:Bridge Infill` is the anchor for your scenario.
<details>
<summary>Screenshot showing "Bridge infill" comment in print preview</summary>

[image]
</details>

You can do this manually using a text editor or creating a post-processing script [1], which I became aware off recently [2].

[1] https://help.prusa3d.com/en/article/post-processing-scripts_283913
[2] https://github.com/prusa3d/PrusaSlicer/issues/7802#issuecomment-1020563954

Kreyren, 2022-01-28:
Seems annoying to do it this way, but i give it a try.

Thanks for taking time to find a possible solution.

closed 2022-01-28 by Kreyren

bubnikv, 2022-01-28:
You may be able to exercise your creativity and riddle solving and to use the regular expression substitution as implemented in the upcoming PrusaSlicer 2.4.1-beta1. The release log will show some examples.