prusa3d/PrusaSlicer #7817: Nozzle Diameter wrong on Support / first layer

cheffeundwackl, 2022-01-26:
### Version
_Version of PrusaSlicer used goes here_
2-4-0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Win 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Marlin 1,1,8

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
_Is this a new feature request?_

I work with a 0.8mm nozzle. False values are used for both the first layer and the support material. See screenshot
[image]
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2022-01-26:
Please post a 3MF.

cheffeundwackl, 2022-01-26:
hope it helps
[Winkel 3 seitig.zip](https://github.com/prusa3d/PrusaSlicer/files/7941654/Winkel.3.seitig.zip)

lukasmatena, 2022-01-26:
I think it's doing what you asked it to do:

[image]

cheffeundwackl, 2022-01-26:
Ok classic, 40cm in front of the monitor is the mistake. 

However, the question arises, why can I adjust the nozzle diameter? Or are there already dynamic nozzles adjustment? Isn’t that a source of error? I can change the flow rate accordingly. 

Thanks for the support

plampix, 2022-01-28:
You can have multiple extruders defined, each with their own nozzle diameter. If you leave the value at 0 for support material, or use a percentage, it will determine the width based on the used extruder.

And a Prusa printer will tell you if there's a mismatch between the nozzle size used to slice the file, and the one currently installed on your printer.

closed 2026-09-29 by prusaslicer-bot[bot]