prusa3d/PrusaSlicer #7818: Freezes

blue68camaro-git, 2022-01-26:
### Version
Version: 2.4.0+arm64
Build: PrusaSlicer-2.4.0+arm64-202112211601
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 12.1 (Build 21C52)
Total RAM size [MB]: 8,590MB

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro V2

### Behavior
Leaving PrusaSlicer open overnight or even a few hours, just sitting in the back ground, you attempt to rotate or open a new model or do anything with the software it is unresponsive. Have to Quit PrusaSlicer and reopen.

n8bot, 2022-01-26:
This reminds me, possibly, of another error that could end up being related in a roundabout way.

Do you have trouble switching to the PS window via the taskbar icon? I only have access to an old macbook, with an older version of macOS, but I think this has been reported to be the case with new versions of macOS.

For me, on the old macbook, when I click the taskbar icon, the window will not be brought up if it's minimized. Instead, I have to hold and select "Show all windows" or whatever the option is. Then, the window will pop up.

So, how is this related to the issue? It's probably not, but I have a far-fetched theory.

Apple likes to cache the image of an app state in order for it to seem to load quicker. I.e., they will capture an image of the app the last time they saw it, so they can show you that picture of the app while the app is loading or not yet available (because the OS is waking up, or something).

It's possible, but not likely, that the "frozen" PrusaSlicer, is just something going wrong with the switching to the window of the app. macOS might be showing you an image being like "yeah, hold on just look at this while we grab that window" but that window never comes.

So, a way to kind of rule out my silly theory here: let PrusaSlicer sit for a few hours to get it to freeze. When it's in its frozen state, hold-click on the taskbar icon and click that menu option to show all windows or whatever. If that does not bring up the window or change the frozen state, I would say my crazy theory has been a waste of time.

kocikdav, 2022-01-28:
This looks like #7645.  @blue68camaro-git please would you send us your current configuration folder? In the other issue, it seems to be caused by switching from 2.3.0-3 to 2.4.0 on same configuration folder. It would be very useful for us, to have such conf folder for testing as we are unable to reproduce on our Mac systems.

bubnikv, 2022-01-28:
> Leaving PrusaSlicer open overnight or even a few hours, just sitting in the back ground, you attempt to rotate or open a new model or do anything with the software it is unresponsive. Have to Quit PrusaSlicer and reopen.

Did your machine fall asleep? Did it hibernate?

blue68camaro-git, 2022-01-29:
> This looks like #7645. @blue68camaro-git please would you send us your current configuration folder? In the other issue, it seems to be caused by switching from 2.3.0-3 to 2.4.0 on same [...]

I will but not sure where it is located and how to attach it........ Can you give me a hint on how too?

closed 2022-01-29 by blue68camaro-git

n8bot, 2022-01-29:
You should have a folder in `~/Library/Application Support/` called PrusaSlicer. Put that entire folder into a .ZIP file and upload it here.

blue68camaro-git, 2022-01-30:
Got it........ I think......

[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/7966216/PrusaSlicer.zip)