prusa3d/PrusaSlicer #7819: PrusaSlicer consider invalid data when two objects are placed and set for vase mode when printing them one at a time

pedgarcia, 2022-01-26:
### Version
Version 2.4.0+arm64
### Operating system type + version

macOS 12.1 on Macbook Pro M1

### 3D printer brand / version + firmware version (if known)

Prusa MKS3+

### Behavior

If you place two objects on the bed and want to print one at a time in vase mode, the slicer consider it as invalid data, although they won't be printed at the same time

So in my opinion, when you have "Vase Mode" and "Complete Individual Objects" set at the same time, it should be able to slice them.

Note: You can reproduce the issue using two cubes from the shape gallery.

rtyr, 2022-01-26:
Please provide your 3MF project file. I am unable to reproduce the issue.

Isn't it just about extruder clearance setting? (you have to increase height clearance for 25mm cube from the gallery)

pedgarcia, 2022-01-27:
Hi,

I had increased the clearance but I believe I've reset it and hadn't noticed. After your message, I reviewed my settings and you are correct about the height. Please, feel free to close this as invalid.

Thanks

Paulo

closed 2022-01-27 by lukasmatena