prusa3d/PrusaSlicer #7821: Bug in Prusaslicer 2.4.0 very fat filling line

Dorito203, 2022-01-26:
### Version
_Version 2.4_

_Version 2.4.0+win64_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Mk3s / irrelevant

### Behavior
At a height of 2.45mm, a very thick fill line is sliced where a normal one already exists. Vojtěch Bubník on Twitter has asked me to provide you with the 3mf for examination.
The whole thing occurs at different layer heights. If you increase the height in 0.01mm steps, this thick line is sliced in some heights and in some not.
I think it has something to do with the model, but fixing it via Netfabb did nothing.

[3mf and Images.zip](https://github.com/prusa3d/PrusaSlicer/files/7944723/3mf.and.Images.zip)

Jebtrix, 2022-01-26:
Looks like github messed up the file upload. Link is invalid.

foreachthing, 2022-01-26:
@Jebtrix file downloaded just fine now.
I took the liberty to post the attached images in this comment...
1. [image]
1. [image]
1. [image]

@Dorito203 please comment these images.

Jebtrix, 2022-01-26:
Link is working now. Even with latest git it does display the behavior. Right before slicing finishes I see the artifact where the issue pops up. The culprit is Detect thin walls.

[image]

Dorito203, 2022-01-26:
Detect thin walls really seems to be the reason. The other walls next to it have exactly the same dimensions and were extruded in the same steps. So I think that some anomaly in the mesh at this location is causing the slicer to create this line.

Jebtrix, 2022-01-26:
The mesh is manifold. There is some intersecting faces but nothing too crazy. The slicer has an issue with the bottom. 

To get it to slice with thinwall correctly either:
- Sink it 0.05mm or cut at 0.5mm to resurface bottom faces
- Merge vertices @ 0.02mm distance
- Redo bottom topology in ie Blender

lukasmatena, 2022-01-27:
Thanks. Reproduced on Windows, not reproduced on Linux or macOS. The bug is already present in 2.3.3. We'll look into it, along with many other issues with thin walls.

bubnikv, 2022-01-28:
The gap fill algorithm uses boost::polygon Voronoi generator, which is not 100% stable, for example 
https://github.com/boostorg/polygon/issues/43

dartrax, 2022-05-01:
May be related to #4495.

closed 2022-05-10 by stepikovo