prusa3d/PrusaSlicer #7825: Editing in G-Code boxes unable to utilize any keys aside from the alphanumeric

hieppo, 2022-01-27:
### Version
2.4.0+linux-x64-GTK3

### Operating system type + version
Ubuntu 18.04.6 LTS

### 3D printer brand / version + firmware version (if known)
Creality Ender3 Pro, Marlin 2.0.1

### Behavior
When editing the G-Code, i.e. Start and End, under Printer preset, The only keys that can use are the alphanumeric keys. None of the standard keys like Enter, Backspace, Cursors can be used. Hence, new lines can't be added. Characters can't be deleted. The only thing can be done to delete is to highlight the area where changes need to be done and overwrite with new entries.

Project file & How to reproduce
Run GTK2 or GTK3 appimage both same results
Click on Printer Settings
Select Custom G-code
Try to add new lines in Start G-code box but fail to do so
Try to delete word but fail as well

Jan-Soustruznik, 2022-08-08:
Hello, @hieppo,  

I tried to reproduce your issue on my Ubuntu 18.04.6 LTS. I cannot reproduce it. Do you use an external or laptop build in keyboard? If you use external keyboard, you could to try it with another one, please.

hieppo, 2022-08-08:
This was posted 1/2 a year ago. The latest version seems to have corrected this issue. I don't know which version that the fix was implemented but it is good now. Bug can be closed. Thank you

FidelCapo, 2022-08-08:
@hieppo Thank you.

closed 2022-08-08 by FidelCapo