prusa3d/PrusaSlicer #7827: 2.4.0 startup issues on ubuntu 21.10 on Framework

GaryAitken, 2022-01-27:
### Version
2.4.0+linux-x64-GTK3

### Operating system type + version
ubuntu 21.10 running on Framework laptop
5.13.0-27-generic x86_64 GNU/linux

### 3D printer brand / version + firmware version (if known)
Prusa I3MK3S

### Behavior
When started, I get the message:
`(prusa-slicer:2626): Gtk-CRITICAL** : 08:48:38:686: gtk_window_resize: assertion 'height > 0' failed`
I then get the "This is the first time you are running prusa-slicer 2.4" message asking if I want to send the configuration.  If I select to send the configuration, the "Sending..." sub-dialog appears and disappears, but the main dialog stays up and I never get any further; I have to kill the process.  Clicking the x in the upper left corner to dismiss the dialog has no affect.  This happens whether using wi-fi or a hard-wired ethernet connection.

The framework has a display which is 2256x1504, but normally runs at 200% magnification, so half that.  I tried running at 100% and get the same messages and problems.

#### Project File (.3MF) where problem occurs
issue happens before any project file is opened.

michnovka, 2022-02-08:
I have the same issue, just with width, not height with Kubuntu 20.04 and PrisaSlicer 2.4.0, both GTK3 and normal.

```
[2022-02-08 15:34:23.576189] [0x00007f69dec928c0] [trace]   Initializing StaticPrintConfigs

(prusa-slicer:4177613): Gtk-CRITICAL **: 15:34:26.049: gtk_window_resize: assertion 'width > 0' failed

(prusa-slicer:4177613): Gdk-ERROR **: 15:34:26.243: The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 738 error_code 2 request_code 151 (GLX) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)
```

hejllukas, 2022-02-08:
> I have the same issue, just with width, not height with Kubuntu 20.04 and PrisaSlicer 2.4.0, both GTK3 and normal.
>
> ```
> [2022-02-08 15:34:23.576189] [0x00007f69dec928c0] [trace]   Initializing [...]

I've seen a similar problem a few times before and also in other applications. It always happened to me after I updated the GPU driver or kernel. At least for me, it always works to restart the computer.

michnovka, 2022-02-09:
@hejllukas yea, a reboot fixed the issue. And yes, I did run kernel as well as nvidia upgrade prior to seeing this. I think we can close the issue

MattBlissett, 2023-01-02:
Same here, and I think #4935 is a duplicate too.

The next startup had this error message:

> *PrusaSlicer started after a crash*
>
> PrusaSlicer crashed last time when attempting to set window position.
> We are sorry for the inconvenience, it unfortunately happens with certain multiple-monitor setups.
> More precise reason for the crash: "*crashed_at_splashscreen_pos*"
> For more information see our GitHub issue tracker: "#2939" and "#5573"
>
> To avoid this problem, consider disabling "*Restore window position on start*" in "Preferences". Otherwise, the application will most likely crash again next time.

However, it was also crashing after I moved my existing configuration in ~/.local/config/PrusaSlicer aside.  Version 2.5.0, both the GTK2 and GTK3 appimages on Ubuntu 22.04.

Zettymaster, 2023-01-06:
I might aswell chime in too.
happened on arch after a system update. pretty sure this was caused by nvidia-open-dkms since i had some other graphical annoyances after the update, not this program in particular.

lukasmatena, 2023-11-22:
Apparently not. Closing.

closed 2023-11-22 by lukasmatena