prusa3d/PrusaSlicer #7829: Initial tool change doesn't use purge tower

StevilKnevil, 2022-01-27:
### Version
2.4.0-win64

### Operating system type + version
Windows 10

### Behavior
When printing with multiple extruders the purge tower isn't used for the first tool change. (Assuming that the purge tower is enabled)

The sliced G-Code has a tool change right at the start, but then goes straight on to print. This means that the filament hasn't been fully purged, spoiling the print. All future tool changes purge correctly.

Either: the start of the print should just use the "current" extruder and not have a tool change and purge. Or the tool should be specified at the start but also purge.

StevilKnevil, 2022-01-28:
I've dug into the code and I realise that previously the purge tower brim would be printed directly after the initial tool change, meaning that the nozzle would be nicely primed before going on to the main print:

```
; generated by PrusaSlicer 2.3.1+win64 on 2021-05-21 at 10:40:15 UTC

...

T0
; Filament gcode
G1 Z0.200 F9000.000
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
M117 INDICATOR-Layer1
G92 E0
;0.2

G1 E-1.50000 F2400.000
G1 Z0.200 F9000.000
;AFTER_LAYER_CHANGE
;0.2
G1 X18.000 Y50.000
G1 E1.50000 F2400.000
;HEIGHT:0.200000
;TYPE:Wipe tower
;WIDTH:0.500000
;-------------------------------------
; CP WIPE TOWER FIRST LAYER BRIM START
G1  X18.000 Y73.500  E1.3397
G1  X15.457 Y73.957   F7000
G1  Y49.543  E1.0207 F2100
G1  X9.543  E0.2473
G1  Y73.957  E1.0207
G1  X15.457  E0.2473
G1  X15.914 Y74.414   F7000
G1  Y49.086  E1.0589 F2100
G1  X9.086  E0.2855
G1  Y74.414  E1.0589
G1  X15.914  E0.2855
G1  X16.371 Y74.871   F7000
G1  Y48.629  E1.0971 F2100
G1  X8.629  E0.3237
G1  Y74.871  E1.0971
G1  X16.371  E0.3237
G1  X16.828 Y75.328   F7000
G1  Y48.172  E1.1353 F2100
G1  X8.172  E0.3619
G1  Y75.328  E1.1353
G1  X16.828  E0.3619
; CP WIPE TOWER FIRST LAYER BRIM END
;-----------------------------------

G1 E-1.05000 F2400.000
;WIPE_START
G1 F7200.000
G1 X16.371 Y74.871 E-0.20470
G1 X15.668 Y74.871 E-0.22280
;WIPE_END
G1 E-0.02250 F2400.000
G1 X98.486 Y108.255 F9000.000
G1 E1.50000 F2400.000
;TYPE:Perimeter
;WIDTH:0.4
;HEIGHT:0.2

; NOTE: MAIN PRINT BEGINS

G1 F1800.000
G1 X98.973 Y107.208 E0.03427
G1 X99.524 Y106.194 E0.03427
G1 X100.137 Y105.216 E0.03427
G1 X100.810 Y104.278 E0.03427
```

however now after the initial tool change command it goes directly to the print.

```
; generated by PrusaSlicer 2.4.0+win64 on 2022-01-27 at 21:31:50 UTC

...

T0
; Filament gcode
G1 Z.2 F9000
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0
;0.2

G1 E-1.5 F2400
G1 Z.2 F9000
;AFTER_LAYER_CHANGE
;0.2
G1 X112.632 Y112.632
G1 E1.5 F2400
;TYPE:Perimeter
;WIDTH:0.45
;HEIGHT:0.2

; NOTE: MAIN PRINT BEGINS
G1 F1800
G1 X117.368 Y112.632 E.1603
G1 X117.368 Y117.368 E.1603
G1 X112.632 Y117.368 E.1603

```

As far as I can tell this change in behaviour was brought about in:
https://github.com/prusa3d/PrusaSlicer/commit/a6ddab85

FidelCapo, 2022-01-28:
Can you please attach zipped 3mf project file? File -> Save project as...
Thank you

StevilKnevil, 2022-01-28:
Here's the project file. I use M600 on tool change to make sure that I can load the correct filament, that happens before the T0 (but after my start-up priming line) and then goes straight on to the main print. Previously it would print the brim around the edge of the purge tower after that initial tool change - perhaps not intentional, but the side effect was that the initial tool-change was primed nicely!

[PurgeTowerBug.zip](https://github.com/prusa3d/PrusaSlicer/files/7960627/PurgeTowerBug.zip)

bombela, 2022-12-14:
Change color at some layer height: initial tool is purged on the wipe tower.
Multi color print: no purge of the initial tool!

It also assumes that de default extruder is always loaded at the beginning of the print. So if you wanted to start the print with T0, and T0 is the default in prusa slicer. But you have T5 loaded right now. It will print with T5... and even dutifully purge it on the wipe tower if it is a layer color change.

StevilKnevil, 2023-03-03:
But that means that the initial tool hasn't been primed as it's (un)loaded before print and then goes straight to attempt to print.

bombela, 2023-03-08:
I designed my own purge system, which is driven by Klipper https://github.com/bombela/EjectionAssistedPurge

Since I do not change material/color often, for example a color change for text at the first/last few layers, a purge system is much faster and simpler. And I can even print a sequence of objects with individual colors this way.

Jan-Soustruznik, 2024-03-11:
Hello, @StevilKnevil,

I see that you using customized start G-code. You should initialize corret print tool before start printing. You have to initialize it by add `T[initial_tool]` before start purge line. Then the correct tool will be used for the printing purge line and no need to neccesary to start print the wipe tower first. 

Closed as solved

closed 2024-03-11 by Jan-Soustruznik