prusa3d/PrusaSlicer #7833: (feature request) In multi-material prints with supports, designate a separate material for non-interfacing support

mmirate, 2022-01-29:
_Is this a new feature request?_
Yes

When using a multi-tool printer to print with a sacrificial support material, the sacrificial material is used for the entire support - tower and interface alike. However, the only material which actually needs to dissolve is the interface. The tower below it only needs to have a similar print temperature. Therefore, unless the sacrificial material is very cheap (e.g. eco-PLA supports for a PETG structure), using sacrificial material for the tower can be a tremendous waste. But at higher temperatures and stranger structural materials (e.g. PCCF and ... HIPS I think?) using either the structural material or the sacrificial material for the support towers would be similarly wasteful compared to using a second structural material of similar print temperature (e.g. ASA) which is much cheaper than either the desired structural material or a sacrificial material of such high print temperature.

Hence, the slicer should have an option to designate a separate material (other than the structural material or the soluble material) for support towers.

n8bot, 2022-01-29:
This is already an option. 

So your printer has three tools? Put the third material in the third tool, and select tool 3 for support material extruder.

mmirate, 2022-01-29:
You're right.

[image]

closed 2022-01-29 by mmirate