prusa3d/PrusaSlicer #7834: Unable to reopen after minimizing

blue68camaro-git, 2022-01-29:
### Version
VersionPrusaSlicer

Version: 2.4.0+arm64
Build: PrusaSlicer-2.4.0+arm64-202112211601

Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 12.1 (Build 21C52)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version: 2.1 Metal - 76.3
Vendor: Apple
Renderer: Apple M1
GLSL version: 1.20

PROBLEM: After minimizing PrusaSlicer to task bar I am unable to maximize. Have to right click, show all windows, select it then it reopens.

blue68camaro-git, 2022-02-01:
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/7981957/PrusaSlicer.zip)

bubnikv, 2022-02-02:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6364
It worked in some older PrusaSlicer version, I believe it was fine in 2.2.0. We need to dig deeper as of what went wrong, whether we broke it or whether it broke in wxWidgets (the UI library PrusaSlicer is built upon).

closed 2022-02-02 by bubnikv