prusa3d/PrusaSlicer #7835: Slicer adding unrecognised commands for reprap firmware

DrJuJu, 2022-01-29:
### Version
Version 2.4.0+win64

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Voron 2.4 Clone 

Board: Duet 3 MB6HC (MB6HC)
DSF Version: 3.3.0
Firmware: RepRapFirmware for Duet 3 MB6HC 3.3 (2021-06-15)

### Behavior
Slicer adds unrecognised G code commands e.g. M107  

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

[First_Layer_Test_v2_0.2mm_PLA_Voron_v2_300_afterburner.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/7963862/First_Layer_Test_v2_0.2mm_PLA_Voron_v2_300_afterburner.gcode.txt)

n8bot, 2022-01-29:
M107 is not unrecognized in rrf, it is deprecated. This means that the use of a different g-code is intended on the newest versions of rrf. 

However, as discussed in a recent issue about this very thing, the new g-code conflicts with g-codes of older versions of rrf. So, it is safer to assume the use of m107. 

M107 still does what it is intended to do, on all versions of rrf.

n8bot, 2022-01-29:
I removed a previous comment I made, which had me confusing M107 and G10... because I had not had coffee yet. Sorry for the confusion.

I suppose that using `M106 S0` is the [documented preferred method](https://duet3d.dozuki.com/Wiki/M107).

DrJuJu, 2022-01-30:
I saw your original comment, and was still confused at to why the slicer should add the M107 command at the beginning of the gcode, even before the start_gcode section ??

And yes I do use the M106 Px Sx to set fans, lights etc.

rtyr, 2022-01-31:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5600. There is already pull request regarding this. Closing.

closed 2022-01-31 by rtyr