prusa3d/PrusaSlicer #7839: Internal error: Calling AppConfig::save() from a worker thread!

cillian64, 2022-01-30:
### Version
2.4.0+win64

### Operating system type + version
Windows 10 Pro, 21H1, OS build 19043.1466

### 3D printer brand / version + firmware version (if known)
Not applicable, I don't get as far as printing anything or setting up settings.

### Behavior
When I close the application or change from simple to advanced mode, a dialog pops up with the message "Unhandled unknown exception: terminating the application.".  After dismissing this, I get a "PrusaSlicer Error" dialog with the message "Internal error: Calling AppConfig::save() from a worker thread!".  Expanding "Details" just shows the same error twice.

Steps to reproduce:
- Fresh install of PrusaSlicer
- Open the application
- Select "Do not send anything" on the telemetry dialog
- Click the yellow "Advanced" button in the top right corner, or alternatively, close the application

Expected results: Not seeing an error and my config being saved

#### Project File (.3MF) where problem occurs
Not applicable, happens when I open the application with no settings and a blank build plate.

#### Other
- I installed the application "only for this user" rather than "for all users".
- This is running in a VirtualBox virtual machine
- This issue appears to be 100% reproducible on my setup.
- When I tried to run the installer "for all users" I got an error saying that the installer was already running.  The installer then appeared to hang until I killed it.  I then tried again installing "only for this user" which succeeded.
- This is running on a system with no network access during install and at runtime.
- I just installed exactly the same version of PrusaSlicer on another Windows 10 Pro 21H1 system and it all works fine
- I found issue #4961 with the same error but given that was on Linux and apparently stopped happening it doesn't sound like this is a dupe.
- I have tried uninstalling and reinstalling PrusaSlicer in my virtual machine and I get exactly the same results.

bubnikv, 2022-01-31:
The check relies on a specific feature: Naming of threads. The threads are named for debugging purposes. I wonder whether your specific issue of running under virtual box would break naming of threads, or whether for some reason PrusaSlicer.ini is really being saved by a worker thread, which would be an error.

Since #4961 we did not ever hear of this issue.

bubnikv, 2022-01-31:
I just tested in a Windows 11 sand box (it is a virtual machine by Microsoft, sharing the system binaries between host and virtual machines) and I was not able to reproduce your issue.

bubnikv, 2022-01-31:
Looks like the issue in #4961 is caused by Linux compatibity layer for BSD, which does not implement thread naming API correctly.

cillian64, 2022-01-31:
It’s pretty weird, the only unusual things in my case are the VM and the lack of network, and I can’t see why either of those would mess up thread naming or cause the wrong thread to try to save.  Lack of network feels to me more likely to be the problem though.

Just wanted to mention that this issue is a a low priority for me so I don’t mind closing if you can’t reproduce it and nobody else ever sees it.

bubnikv, 2022-02-01:
Our tester @FidelCapo tried to reproduce with disconnected network with no success.

cillian64, 2022-02-11:
Ok, since you can't reproduce this and I'm not using the troublesome VM any more I'm happy to close this issue and forget about it.  Thanks for your help and debugging :-)

closed 2022-02-11 by cillian64

bubnikv, 2022-04-22:
No issue running https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc2 in Windows Sandbox (it is a kind of virtual machine). I asked @YuSanka to test on her Windows 7 virtual machine.

cirk2, 2022-09-11:
I encounter this issue with the current release 2.5.0 running it in a win10 VM using VirtualBox 6.1.

[image]

It seems like the main thread renaming fails under Virtual Box.

glenstewart, 2022-11-13:
I stumbled into the same problem, running Virtual Box 6.1 under Linux Mint Vanessa.  The fix was to turn off 3D acceleration.

kohend, 2022-11-22:
After disabling 3d support it starts, but it fails on the same thing when trying to save a file...
[image]

bubnikv, 2023-02-07:
Hopefully fixed with 84722876012ca310e29b291e10fef9d18ae26cea
    Implementing a test whether the current thread is the main (UI) thread
    and using it on AppConfig::save() to assert if save is called
    from a worker thread.
    The old assert was using thread names, which did not always work
    on Windows.

MichaelVanEnde, 2023-02-09:
I encountered the same isseu today. And i'm running native windows 11

Guenni75, 2023-02-10:
I have the same Problem if i close the Slicer after i worked with it.
If i just open and close, its ok.

I think it is the "last Project" list.
The List hasn't changed since the error!
Win10, 2.5.0, 64bit

[image]

niklasmeyn, 2023-02-15:
Hi. In my case, I could solve the problem by selecting the graphic card within MS. It seems somewhat related to NVIDIA
[image]

SweAcMe, 2023-02-22:
> Hi. In my case, I could solve the problem by selecting the graphic card within MS. It seems somewhat related to NVIDIA [image]

Solved. I've the same hw as niklasmeyn. After choosing Intel GPU the error is gone!
/SweAcMe

lukasmatena, 2023-02-22:
@SweAcMe The problem should already be fixed in 2.6.0-alpha4. Can you please confirm that?

niklasmeyn, 2023-02-22:
> @SweAcMe The problem should already be fixed in 2.6.0-alpha4. Can you please confirm that?

@lukasmatena I can confirm that the issue is no longer present in the 2.6.0-alpha4. My issue was previously experienced with 2.6.0-alpha3

lukasmatena, 2023-02-22:
@niklasmeyn Thanks for the update.

SweAcMe, 2023-02-22:
> > @SweAcMe The problem should already be fixed in 2.6.0-alpha4. Can you please confirm that?
> 
> @lukasmatena I can confirm that the issue is no longer present in the 2.6.0-alpha4. My issue was previously experienced with 2.6.0-alpha3

Confirmed!

ashtricky, 2023-02-27:
Dell precision 14" w/ nvidia gpu - same error - this worked if I forced it over to integrated graphics.

codemonkeycodes, 2023-03-07:
Latest alpha build fixed this issue for me.  Moved to a new PC, launching 2.5.0 crashes on startup, but latest 2.6.0 alpha (as of 3/7/2023) launches as expected.  The workaround for 2.5 to change to to run on the integrated video card lets the application launch.

alexw-im, 2023-04-06:
I'm now experiencing this same issue with a vanilla windows 10 install on an HP Zbook Fury G8, only after updating to 2.5.2. I now can't launch PS at all.

Going to try upgrading to the 2.6 alpha but that's not ideal for its own reasons.

gshick, 2023-04-28:
Just adding another comment for same issue.  **Internal error: Calling AppConfig::save() from a worker thread!**

Windows 11 pro on brand new Dell Inspirion.  Have tried adjusting NVIDIA graphics settings to no avail.

Going to try 2.6 alpha next.

jeff-o, 2023-07-19:
I can confirm that using the Windows graphics settings (as SweAcMe commented) fixed the issue for me. Using the Nvidia Control Panel to force integrated graphics does _not_ work. 2.4.58.5.

AmshaSharam, 2023-08-03:
I also get the same error.
Win 10
2 grafics, intel+nvidia
SuperSlicer : Nightly windows x64 2.5.59.2
i will try change grafic.

lukasmatena, 2023-08-03:
@AmshaSharam We are not interested in SuperSlicer's bugs, unless they are also reproducible in PrusaSlicer 2.6.0.

jeff-o, 2023-08-05:
> @AmshaSharam We are not interested in SuperSlicer's bugs, unless they are also reproducible in PrusaSlicer 2.6.0.

In my case, it affected both Superslicer and Prusa Slicer (version 2.5.2). I just installed PS 2.6.0 and I do not see the same error.