prusa3d/PrusaSlicer #7840: S

bramp, 2022-01-30:
### Version

Version: 2.4.0+x64
Build: PrusaSlicer-2.4.0+x64-202112211607

### Operating system type + version

Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 12.2 (Build 21D49)
Total RAM size [MB]: 34,360MB

### 3D printer brand / version + firmware version (if known)

Prusa MK3S+ 

### Behavior

**Describe the problem**

PrusaSlicer will crash when I try and change my Configuration. 

**Steps needed to reproduce the problem**

1. Start PursaSlicer
2. Click Configuration menu, then Configuration Assistant.
3. With or without changing anything, click "Finish"
4. PrusaSlicer crashes.

**Expected Results**

1. It not crash :)

**Actual Results**

It crashes with no GUI error message. If I start Pursa from the command line, I see `Segmentation fault: 11`:

```
$ /.../PrusaSlicer.app/Contents/MacOS/PrusaSlicer
[2022-01-30 11:43:01.180419] [0x000000011f6b9600] [trace]   Initializing StaticPrintConfigs
[2022-01-30 11:43:01.623448] [0x000000011f6b9600] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file)
Segmentation fault: 11
```

I can reliability repeat this, and this is currently stopping me from adding new filaments :(

FidelCapo, 2022-01-31:
I think it's a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7655#
This bug should be fixed by now and the fix will be in version 2.4.1-beta which should be out in the next few days

closed 2022-01-31 by FidelCapo

bramp, 2022-01-31:
Thanks! I should have more thoroughly searched for existing duplicate issues.