prusa3d/PrusaSlicer #7842: [Bug] Bed temperature drops to 60C in the middle of the print even I specified otherwise

laptop-laptop, 2022-01-30:
### Version
2.4

_Use `About->About PrusaSlicer` for release versions_
2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
* _Describe the problem_

I want to have bed temperature at 75C throughout the entire print process. So I went to Filament>Filament Settings>Temperature>Bed to set both First layer and Other layers temperatures to 75C. However, during the printing processing, I noticed that on the LCD the bed temperature dropped to 60C. It should be 75C as I specified in 'Other layers'.

rtyr, 2022-01-30:
Please provide your 3mf project file and the gcode.

rtyr, 2022-03-30:
No response. Closing.

closed 2022-03-30 by rtyr