prusa3d/PrusaSlicer #7843: Gcode viewer not rendering G60/G61...

jmerc77, 2022-01-31:
### Version
Version 2.4.0+win64

### Operating system type + version
PrusaSlicer G-code Viewer
Version:   2.4.0+win64
Build:     PrusaSlicer G-code Viewer-2.4.0+win64-UNKNOWN

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 17,121MB
OpenGL installation
GL version:   4.6.0 NVIDIA 511.23
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1660 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 341 lines cut ...]
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### Behavior
* G60/G61 should be rendered as the save/load positions like marlin actually prints it.
ex:
G1 X20 Y20;could be any move...
G60
G1 X10 Y10
M0
G61
G1 X30 Y30
* should (in the example) save location (X20, Y20), move to X10 Y10, pause, move to saved location (X20, Y20), then move to X30 Y30 in preview
* moves to X20 Y20, moves to X10 Y10, then moves to X30 Y30 in preview (it's skipping the G61)

#### Project File (.3MF) where problem occurs
in my case, occurs during my pause...
[test1.zip](https://github.com/prusa3d/PrusaSlicer/files/7968197/test1.zip)

rtyr, 2022-01-31:
https://marlinfw.org/docs/gcode/G060.html

lukasmatena, 2022-05-27:
> I tried with current master (2.6.0-alpha0, ish), but it complains that my file (above) does not contain valid gcode.

The gcode snippet is too short for the G-Code Viewer to make sense of it. If you generate some simple gcode in PrusaSlicer, it will gladly open it. Then you can add the G60/G61 and check whether they are shown correctly. In current master, they should.

closed 2026-09-29 by prusaslicer-bot[bot]