prusa3d/PrusaSlicer #7846: Fix incorrect handling of G92 E0

n8bot, 2022-01-31:
@bubnikv Please consider this. Unnecessary g-code commands and incorrect error messages are not useful!

In commit "Fix of #6336 #5073: Throw an error if G92 E0 is missing in layer change" 58d64bae7762e225a9755a79c80068f56a81b5f5

An error message will be displayed if the user does not use `G92 E0` while using relative extruder moves (but not with absolute extruder moves).

This is the exact opposite of useful. It is **absolute** extruder moves that need a `G92 E0`. Relative moves absolutely do not need it.

There are two problems that are being addressed here:

1. **Print errors**
1. **PS Visualization errors**

The first case, print errors, ONLY appears if **absolute extrusion** is used. There is no need whatsoever to use `G92 E0` with relative extrusion for a print. Every move is starting from zero.

The second case, PS visualization errors, can be remedied with a `G92 E0`, yes. **However***, it is better to be fixed internally instead of relying on users having a `G92 E0` in their layer change g-code. See this fix: https://github.com/n8bot/PrusaSlicer/commit/2fd8ca2a1869597c88d69ca13db320ff9163642e

bubnikv, 2022-02-01:
>    2. *PS Visualization errors*

> Relative moves absolutely do not need it.

You are not correct. At least for Marlin, what you see on the screen in
PrusaSlicer simulation mode is what you will get printed on the printer.

n8bot, 2022-02-01:
Explain to me how what you are saying makes sense. Relative moves always originate from 0. Thus, there is no accumulation of floating point error.

Absolute moves reference an imaginary absolute E axis... thus, they need to be reset.

I think you have things backwards.

n8bot, 2022-02-01:
Related: https://github.com/prusa3d/PrusaSlicer/issues/4379#issuecomment-646182202

In which, the "answer" provided is this:

> The firmware internally maintains an absolute counter even if it is addressed relatively in the g-code. This G92 E0 resets the internal counter, thus preventing the accumulation of errors.

But for what purpose is that internal tracking? Print statistics, etc. Not for printing -- this accumulation error internal to the firmware does not and should not affect a print.

rtyr, 2022-02-01:
>  It is absolute extruder moves that need a G92 E0.

You certainly cannot use `G92 E0` command in custom before/after layer change g-code with absolute mode at the moment.

Some example what would happen (extrusion of over 204mm instead of retraction).

[image]

Therefore an error message to prevent the creation of such unusable g-code is needed.

n8bot, 2022-02-01:
I see. Thank you for bringing that to my attention. That seems to be a problem with PrusaSlicer. Absolute extrusion is typically used in combination with `G92 E0`, yes, even on Marlin.

So, I guess that second part of the error message makes sense, but I was more referring to this part:

> "Relative extruder addressing requires resetting the extruder position at each layer to prevent loss of floating point accuracy. Add \"G92 E0\" to layer_gcode.";

I do not see why these error messages are the easiest solution. There seems to be a misunderstanding on your part, or some firmware/slicer problems that are much more difficult to solve than they seem on the surface.

bubnikv, 2022-02-01:
You should read the firmware source code to understand the issue before you give out categorical statrements. Indeed, the issue was found by me 6 years ago on our print farm when we switched from absolute to relative extrusion and Marlin started to mangle the prints with full print bed.

lukasmatena, 2022-02-01:
Quoting from https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/gcode/gcode.cpp:
```
/**
 * Set XYZ...E destination and feedrate from the current GCode command
 *
 *  - Set destination from included axis codes
 *  - Set to current for missing axis codes
 *  - Set the feedrate, if included
 */
void GcodeSuite::get_destination_from_command() {

(commands unrelated to extruder axis)

  #if HAS_EXTRUDERS
    // Get new E position, whether absolute or relative
    if ( (seen.e = parser.seenval('E')) ) {
      const float v = parser.value_axis_units(E_AXIS);
      destination.e = axis_is_relative(E_AXIS) ? current_position.e + v : v;
    }
    else
      destination.e = current_position.e;
  #endif
```

Although I never worked on firmware and am not familiar with the Marlin codebase, it seems that the position is always tracked in absolute coordinates internally. In case the mode is absolute, the value calculated on slicer's side is taken directly from GCode. If the mode is relative, it is incrementally added to the desired position, leaving room for error accumulation.

My conclusion is that relative mode is prone to error accumulation, absolute is not (unless it happens on slicer's side). Which is surprisingly in accordance with what @bubnikv (who worked on MK2 firmware in the past) stated.

> There seems to be a misunderstanding on your part

Enlighten us then. What is the reason for inaccurate printing with relative extrusion mode on long prints when `G92 E0` is not periodically called?

bubnikv, 2022-02-01:
Verified on Marlin 2 source code base. The accumulators are 32bit floats , the same as it used to be 6 years ago on 8 bit AVR based controllers. The overflow will happen.

> My conclusion is that relative mode is prone to error accumulation, absolute is not (unless it happens on slicer's side). Which is surprisingly in accordance with what @bubnikv (who worked on MK2 firmware in the past) stated.

Absolute mode is prone to overflow indeed, however Slic3r always used to reset the extruder (it emits G92 E0) after each retraction (or deretraction, I don't remember, but it does not matter).

Other firmwares may not have this kind of issue, for example Klipper likely accumulates in 64bit doubles, because it runs in Python which only knows a 64bit double IMHO.

I will check @n8bot beloved RepRapFirmware.

bubnikv, 2022-02-01:
Indeed, the firmware must keep the absolute counter, because one may want to switch the firmware from relative to absolute mode and vice versa, thus the firmware needs to remember the absolute position to be able to do that.

n8bot, 2022-02-01:
> I will check @n8bot beloved RepRapFirmware.

Don't worry. I have checked. 

Indeed, the total extrusion amount is accumulated in a single precision float --  it this doesn't matter! This doesn't affect print quality. The accumulated value is only used for print statistics. The raw value sent in g-code is used to prepare the steps for the next move. This makes sense. Any other approach is pretty silly. 

I have never heard a marlin user say they needed a 'G92 E0' to fix any error.

What you are saying and doing here goes against all conventional wisdom. 

Are you sure that your marlin firmware was not referring to absolute extrusion when you were testing this years ago?

When marlin is commanded to 'G90' absolute coordinates mode, it will also automatically switch to absolute extrusion mode. One has to set 'M83' again after to stay in relative extrusion mode. 

In this case, certainly, you'd need G92 

If you look in the code posted here from marlin I guess, it shows exactly what I mean. 

The firmware uses the value directly from gcode if possible. If there is no value, it uses the current accumulated value. (Why would there be no value?)

And in regards to switching back and forth from relative extrusion to absolute extrusion: This makes not very much sense. In fact this makes zero sense.

If the print is sliced on absolute units, it must stay in absolute units. If the print is slides in relative, it must stay in relative. There is no going back and forth. 

The only need to go back and forth is to do special purging moves, or tool changes and the like. If the user, for example, uses absolute coordinates for the print, then switches to relative for those special moves, there is no need to track the relative moves to be added to the accumulated absolute total. The slicer expects the accumulated total bot to change. 

You have convinced me further that my assessment here is correct.

n8bot, 2022-02-01:
>  (converting to absolute E on G-code parser, then calculating the E difference before passing it to planner).

Are you sure that modern non-prusa Marlin still does this? I tried to look, and it seems like it may still do it.

Sorry for the misunderstanding. TBH, though, it seems silly that marlin would have ever done it this way.

dc42, 2022-02-01:
@bubnikv, I guess I don't fully understand why Marlin needs that G92 E0 when using relative E coordinates; although I can see why it might need to see a G92 E0 command at regular intervals if it relies on an internal relative-to-absolute accumulator to get its calculations right.

n8bot, 2022-02-02:
Sorry for all the misunderstanding and confusion. 

I dug deeper into the Marlin code than I have ever explored. It's certainly different than what I am used to seeing in RRF. It certainly seems to use the accumulated 32 bit float value to generate steps.

In the future, I'll try to be more specific in my wording of issues so we can hone in on what exactly is at play. This issue is clearly only related to RRF. I shouldn't assume that Marlin, or any other firmware, handles things the way I am used to.

lukasmatena, 2022-02-17:
The check is now only active for Marlin firmware. The G-Code preview now uses higher precision and therefore does not visualize the problem. Closing.

commit 3e9e3fd1 (2022-02-14): Follow-up to 58d64bae7762e225a9755a79c80068f56a81b5f5
Disabled check for "G92 E0" in layer change G-code for other firmware types
than Marlin 2 and Marlin Legacy. We don't know whether other firmware
flavors are sensitive to E axis accumulation error and we know that
RepRapFirmware is not.

Fixes #7846
closed 2022-02-17 by lukasmatena