prusa3d/PrusaSlicer #7848: Not error, just an idea - When slicing, I shoud be able to change first layer color, not just that I marked as color change

DKotekCZ, 2022-02-01:
Not error, just an idea - When slicing, I shoud be able to change first layer color, not just that I marked as color change

neophyl, 2022-02-01:
You can, just not in the same place. There’s already a issue here about it (or 2) but you can either set the colour for the filament in the filament profile which is where the initial colour comes from or the easier way is on the plater, right hand panel the filament drop down, the coloured square next to the filament name, click on it and the colour picker should appear.

DKotekCZ, 2022-02-02:
I use another workaround... but nevermind.
I've another 2 issues on my Prusa MK3s+:
1. If you start print with fillament change, you'll get remaining time to a change, that's cool. But if you cancel print and start print without fillament change, it keeps showing information from previous print.
2. If you start print and you cancel print while after temps ares ok and Z-axis starts decreasing, hole software will go mat.

rtyr, 2022-02-02:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6985.

Other issues you mentioned are firmware related, please open issue at https://github.com/prusa3D/Prusa-firmware.

Closing.

closed 2022-02-02 by rtyr