prusa3d/PrusaSlicer #7849: "Save Project" command doesn't work properly

CuredPrusa, 2022-02-01:
### Version
Version 2.4.0+win64

### Operating system type + version
Build: PrusaSlicer-2.4.0+win64-202112211614
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,287MB

### 3D printer brand / version + firmware version (if known)
doesn't matter (Creality CR-10 V3)

### Behavior
* _Describe the problem_
I use always use "Project" (.3mf) when working on my projects. Project instantly loads all needed profiles and 3D object, so it's convenient. But I can't figure out when/why Ctrl+S gets greyed out. I change something in settings, I save the settings, but next time I don't get those new values. I have to recall them by “undo arrow” (next to the value). If I want to really save the changes in the project, I actually have to use SaveAs command and overwrite the file. Ctrl+S doesn't work. 
* _Steps needed to reproduce the problem_
Create two different Projects.
Change some settings and you'll see CTRL+S stays grey. 
Save those settings.
Toggle Projects.
New settings are overwritten by old settings from the project.  
* _Expected Results_
Have Ctrl+S always available for safe and reliable project saving.  
It would be beneficial that project saving automatically saves changes in Settings profiles, because if I want to save something that means I'm satisfied with changes - changed values are verified and I want to save them.   
* _Actual Results_
Tedious workaroud with SaveAs, or lost changes in Settings profiles related to Project
_Is this a new feature request?_
I dont know. It should be a normal command.

#### Project File (.3MF) where problem occurs
doesn't matter

Jebtrix, 2022-02-02:
Looks to be fixed in latest, try [2.4.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1)

CuredPrusa, 2022-02-02:
I have just tried new 2.4.1-beta1.
The problem remains...
Scroll through screenshots (by numeric order) to see the workflow. Observe how Settings are NOT modified (upper right corner), because they WERE modified and properly saved individually.  However SAVE (Ctrl+S)  is disabled (grey). After recalling the same project back I receive a modification again as if it was unsaved. When I press undo arrow (next to the value), I get the correct value which I actually saved before. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jebtrix, 2022-02-02:
OK, I had to distill this issue a bit more. Only one project is needed to demonstrate the bug that is indeed in both v2.4.0 and v2.4.1-beta1.

If you open a project, make setting changes and save the preset with the _**same**_ preset name it clears the project modified flag. This disables Project Save. Next time you open that project its using older embedded preset values because of course, you didn't/couldn't save it.

[image]

CuredPrusa, 2022-02-04:
Nice demo. You nailed it. But it's not only about settings. When I change postprocessing code on Z scale, there is no Save available afterwards.  I'm new here, so tell me please, is communication with you enough to expect some improvements in next version, or should i escalate this further (how?) ?

Jebtrix, 2022-02-04:
Well custom script/gcode is still a setting. If you add/modify any text without leaving the textbox then yes Save Project is unavailable. Just hit Tab to leave/change focus to next control or click on something like any whitespace on the left side panel. It wants to know your done editing before it checks for modifications.

Devs read all the issues, just be patient.

Just be patient, the devs

Jebtrix, 2022-02-05:
If you want a slightly cheesy workaround in the mean time just add a blank space or new line at the end of any Notes textbox to trigger a project modified so you can save.

lukasmatena, 2022-02-17:
This should be fixed in 2.4.1-beta2. Closing.

closed 2022-02-17 by lukasmatena