prusa3d/PrusaSlicer #7850: [FR] G-Code Substitutions: move [X Delete all] button or ask for confirmation

foreachthing, 2022-02-01:
### Version
2.4.1b1

### Operating system type + version
Windows 11

### Behavior
* _Behavior_
  * Want to [+ Add] new G-Code sub,
  * moved the mouse too fast, click came late and [X Delete all] was pressed!
* _Expected Results_
  * I want to be asked, if I was sure to delete _all_!
  * Option: move the [X Delete all] button further away from the [+ Add] button.
* _Actual Results_
  * all regexes gone to the last profile saved.

_Is this a new feature request?_
Yep. 
But that's not an easy request for me, since I hate to click more than necessary...

lukasmatena, 2022-02-03:
@foreachthing A confirmation dialog was added for 2.4.1-rc. Closing.

closed 2022-02-03 by lukasmatena

foreachthing, 2022-02-03:
@lukasmatena, thanks a bunch! 👍🏼