prusa3d/PrusaSlicer #7852: Compare Filament Dialog: Values are truncated

aclucca, 2022-02-01:
### Version
Version: 2.4.1-beta1+win64
Build: PrusaSlicer-2.4.1-beta1+win64-202202010918

### Operating system type + version
Betriebsystemname	Microsoft Windows 10 Pro
Version	10.0.19044 Build 19044

### Behavior
Compare Filament Dialog: Values are truncated - see screenshot

[image]

Jebtrix, 2022-02-02:
If you right click on that row it will show you full comparison dialog.

aclucca, 2022-02-02:
ok, thank you

so . .  maybe a hint will be enough . .

bubnikv, 2022-02-02:
see the 2.4.1-beta1 change log 
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1

The "Unsaved Changes dialog" used to inform that some of the differing keys are too long to be displayed fully and that one may open a pop-up window to show the full content of the differing keys. This was broken in 2.4.0 and it works again now #7734.

closed 2022-02-02 by bubnikv

aclucca, 2022-02-02:
this is a error in the beta version:

Version: 2.4.1-beta1+win64
Build: PrusaSlicer-2.4.1-beta1+win64-202202010918

unsaved dialog is fixed, but not the filament compare dialog