prusa3d/PrusaSlicer #7853: Linear Advance for External perimeters only

ahmetcemturan, 2022-02-02:
### Version
Feature request

It would be great if we could limit the use of Linear Advance to External perimeters only (Including Bridges, Small perimeters etc.)

It would greatly reduce extruder noise, extruder wear and print time.  As Linear advance is mostly an esthetical thing it makes no sense applying it to infill and inner perimeters.

plampix, 2022-02-02:
That's actually a nice use case for the new search&replace code. If your firmware has EXTRA_LIN_ADVANCE_K enabled, you can switch between two profiles using M900 S0 (the value set via M900 K), or M900 S1 (the value set via M900 L).

I haven't tried printing this yet, but I will.

[image]

plampix, 2022-02-03:
Another solution would be adding the linear advance value in the filament preferences, instead of changing the start gcode, and make it available using a placeholder. But I don't know if the search&replace feature allows you to use a PrusaSlicer placeholder like that.

wavexx, 2022-07-09:
IMHO it still makes sense to use LA for infill, although it can be lowered. This is because infill-to-perimeter overlap is larger than regular perimeter overlap, and can bleed into the outer perimeter in several circumstances.

I like SuperSlicer's approach here, where LA can be changed per-feature. The issue here is that this is a per-filament setting, but it does affect quality. One of those eternal debases as to where this setting should be... filament or print profile...

If the filament g-code would be evaluated on each feature then we could do this through a series of if switches directly in the custom gcode without adding more features.

In that case we would just re-emit the LA factor via M900 K instead of switching with M900 L. I suspect the gcode size difference would be negligible and it's less prone to mistakes IMHO since there are no hidden values stored.

djrampage, 2022-07-26:
i made a little modification of this substitutions bc i have issues with solid infill.
```
;TYPE:(.+? infill)
${0}\nM900 K0
Disable LA for any infill

;TYPE:(?!.+? infill).*
${0}\nM900 K0.05
Enable LA for everything else
```
sliced g-code shows that m900 command inserted properly

awabom, 2023-04-22:
Until the feature is implemented, I've made a script that inserts 

M900 K0 

before all 'internal infill' and then restores any previously found M900-command when the infill is done. So it works automatically for different filament types with different custom g-codes.

https://github.com/awabom/3dprint/tree/main/PrusaSlicer-InfillNoLA

closed 2026-09-29 by prusaslicer-bot[bot]