prusa3d/PrusaSlicer #7855: Better infill for internal bridging?

ulab, 2022-02-02:
### Version
2.4.0

### Behavior
I am trying to print models with very low to no infill, but if they have internal structures, those often fail to print because there is no support for internal bridges.

[image]

There are of course workarounds to add modifiers or layer based higher infill, but it would be a lot nicer if these inside structures could have something similar to the outside supports. They are not even supported by "Support cubic" infill - see above.

_Is this a new feature request?_
Yes.

OopAck, 2022-02-02:
Agree

Alpacad, 2022-02-02:
@ulab 
I saw from the commits that they are working con Cura's _**lightning infill**_, which pourpose is to create high quality top surfaces using the least amount of material. Maybe that's what you're looking for.
I suppose it will be implemented in future **2.5.0 release**.

[image]

ulab, 2022-02-02:
I have seen that, but I do think it's only for the top layers too, isn't it?

That said, can the lightning infill be combined with regular infill? You might need some infill in the part to strengthen the print and something like lightning for top layers.

Alpacad, 2022-02-02:
You mean they wouldn't support the internal surfaces? They should be considered top layers too.

To combine different support structures you can use modifiers. Depends on what you want to achieve, but for example by using height range modifiers you can use **_adaptive cubic_** in the middle and **_lightning infill_** on the top.

lukasmatena, 2022-02-02:
Duplicate of #469.

ulab, 2022-02-02:
True, did not find that, thanks.

closed 2022-02-02 by ulab