prusa3d/PrusaSlicer #7856: Grey square rendered over "notebook tabs" in preferences dialog

n8bot, 2022-02-02:
On latest master, I noticed this:

[image]

It goes away as soon as it is hovered over.

I spent a lot of time trying to figure out which commit does it, but I can't seem to find it. I suspect a change in the deps might have caused this.

It doesn't seem present in the 2.4.1-beta1 that is downloaded from GH. Only when I build master myself.

n8bot, 2022-02-03:
I think I ruled out any changes in the deps. There were very few commits, none of which seemed to make a difference to the weird square being rendered. I'll continue exploring. 

Is this reproducible to anyone else?

bubnikv, 2022-02-03:
I see it on my computer

FidelCapo, 2022-05-09:
This issue should be fixed in master.
Closing.

closed 2022-05-09 by FidelCapo