prusa3d/PrusaSlicer #7857: Highlited buttons in language selection menu

Alpacad, 2022-02-02:
### Version
Version 2.4.1-beta1+win64

### Operating system type + version
Windows 10 Pro 21H2 / Windows 11 Home 21H2

### Behavior
"OK" buttons in language selection are highlited on launch.
- Open PrusaSlicer
- Select Configuration > Language
- See the firts highlited button, press OK to continue
- See the second highlited button

#### Attached images
[image]
[image]

n8bot, 2022-02-02:
This is intended, for tab traversal.

bubnikv, 2022-02-03:
See the comment from https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1

> Windows specific: Button in focus and the default button was not highlighted. The issue was introduced with the dark mode support in PrusaSlicer 2.4.0-beta1. Because of missing Win32 dark mode support for buttons, we had to resort to flat rendering of buttons. However with flat rendering, Win32 buttons do not support highlight of a default or active button and we had to implement a workaround ourselves https://github.com/prusa3d/PrusaSlicer/issues/7564.

We know it is not ideal, we hope to improve for 2.5.0.

Alpacad, 2022-02-03:
Understood, thank you.
I've read the changelog pretty quickly and I missed the Bugs Fixed section, sorry.

Closing.

closed 2022-02-03 by Alpacad

n8bot, 2022-02-03:
Notepad++ seems to find a workaround for the dark mode buttons -- their buttons have borders, appear dark themed, and have focus marking.

Are they using purely custom painted controls for the buttons?