prusa3d/PrusaSlicer #7860: [feature request] random part placement (not always center) textured sheet wear levelling

cnlson, 2022-02-03:
### Version
_Version of PrusaSlicer used goes here_2.4.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ w10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_none

### 3D printer brand / version + firmware version (if known) mk3s mostly plus
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
mk3s mostly plus
### Behavior
* _Describe the problem_ in 4 years i have had 3 PEI plates worn out in the center. The original pei sheets and 2 of the textured. 
when all of the printing is always centered , the center gets most of the wear. I know I can manually move the parts but if there are multiple parts and I hit arrange they all end up in the middle again. I would like the arrange function to choose different places on the sheet in a pseudo random way to make the sheet get used in full. 

or say 4 quadrants of 120x105 and center the parts inside and if the amount of coverage exceeds, then moved to the next available quadrant. then for the next print 3rd quad, and subsequent 4th and then wrap. travel speed is 200mm so there would be minimal time difference if you were spanning 4 back to 1 (since they should always be adjacent.)

      2  |   3
 ------------
      1  |   4
  

_Is this a new feature request?_ YES

lukasmatena, 2022-02-03:
Duplicate of #320. Closing.

closed 2022-02-03 by lukasmatena

cnlson, 2022-02-03:
@lukasmatena ok, if this has been a request since 320, is there no chance of it happening?

bubnikv, 2022-02-03:
We are considering the feature, however there is so much on our plates.

cnlson, 2022-02-03:
You need bigger plates. 😂