prusa3d/PrusaSlicer #7864: Infill/Perimeters Overlap Weird Behavior

sarf2k4, 2022-02-03:
### Version
2.4.0+win64

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### 3D printer brand / version + firmware version (if known)
MK3sMMU2s + 3.9.1

### Behavior
Adding a "part modifier" > ["infill/perimeters Overlap" + "Layers and Perimeters"] for an object resulting a weird slicer behavior where the stock value of "infill/perimeter overlap" at 25% didn't produce any perimeters, while other than 25% produces doubled the perimeters on the modified area.

###Expectation
I am expecting this is not inconsistent, and I wasn't sure which one is the actual behavior, should it have another extra parameters or not, I am kind of like the idea of having both not having perimeters and extra perimeters over the modified area for extra rigidity and aesthetics.

###Proposal
Add few extra option for Infill/perimeter overlap section:

- Produce perimeters over modified area [Boolean]. False being the 25%, true being 20% on the image below
- Perimeter types [Both, Outside, Inside] of the modifier. Option [both] would results in the 20% on the image below

#### Project File (.3MF) where problem occurs
[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/7992761/Shape-Box.zip)
[image]

bubnikv, 2022-02-03:
The infill perimeter overlap is part of the perimeter settings, thus the current slicer behavior is to insert new perimeters if the overlap of the modifier differs from the object it modifies. It may not be ideal, it is just a technical explanation of what is going on.

sarf2k4, 2022-02-03:
> The infill perimeter overlap is part of the perimeter settings, thus the current slicer behavior is to insert new perimeters if the overlap of the modifier differs from the object it modifies. It may not be ideal, it is just a technical explanation of what is going on.

Ahhh.... I see it now. thank you for clarifying on this, I almost never really played with this options

closed 2022-02-03 by sarf2k4