prusa3d/PrusaSlicer #7865: Output filename parameters. Month and Day should be 2 digits

Antynik, 2022-02-03:
PrusaSlicer 2.4.1-beta1

This:
[year][month][day]
result in this:
202223
When it should be:
20220203

repeats issue 4176
https://github.com/prusa3d/PrusaSlicer/issues/4176
solved with comment "It will be released with PrusaSlicer 2.4.0-beta2"
but the error is still there

murk-sy, 2022-02-03:
You have to use `zdigits` as noted in that issue https://github.com/prusa3d/PrusaSlicer/issues/4176#issuecomment-954677457

`{zdigits(year, 2)}-{zdigits(month,2)}-{zdigits(day,2)}` produces `2022-02-03`

`{year}-{month}-{day}` produces `2022-2-3`

Alternatively you can use `{timestamp}` which produces `20220203-114839`

I would prefer it to be default as well, but that could break backwards compatibility for some.
Ideally something like `{isodate}` which produces `2022-02-03` would be added.

Antynik, 2022-02-03:
OK. Maybe need to write about it in examples

rtyr, 2022-02-04:
I believe we can close this.

closed 2022-02-04 by rtyr