prusa3d/PrusaSlicer #7870: Add support for slicer version placeholder output file name

Hologos, 2022-02-03:
### Version

Version 2.4.0

### Behavior

Since every new version of PrusaSlicer brings new features, fixes bug and introduces new ones :-) it would be great if there is a slicer version output file name placeholder. This way, I could generate gcode from the same project with different versions of slicer and compare the results.

_Is this a new feature request?_ Yes

neophyl, 2022-02-03:
Is the one at the top in the actual generated gcode not sufficient to do this when comparing files ?

Hologos, 2022-02-04:
Hi, I meant to compare generated gcode in G-code Viewer (visually, such as seam placement).

lukasmatena, 2022-02-04:
@Hologos There IS a placeholder for PrusaSlicer version: `version`:
[image]

That should be it, right?

Hologos, 2022-02-04:
Hi, it was not listed in [list of placesholders](https://help.prusa3d.com/en/article/list-of-placeholders_205643), I dared to try. It's now on the website so you can close this issue. Thank you

bubnikv, 2022-02-04:
thanks, closing

closed 2022-02-04 by bubnikv

lukasmatena, 2022-02-04:
@Hologos There was a lot of things not listed. Hopefully the documentation is up-to-date now.