prusa3d/PrusaSlicer #7871: Using 3DConnexion space mouse goes wrong

sturmth, 2022-02-03:
### Version
_2.4.0+win64_

### Operating system type + version
_Windows10_
_All actual drivers for system and hardware_

### 3D printer brand / version + firmware version (if known)
_n.a._

### Behavior
_Use of space mouse from 3DConnexion was supported on any earlier versions of PrusaSlicer.
After starting the app it still runs fine and you can use the cad mouse in any rotation and translation. But when You tick any menu or dialog on PrusaSlicer it is over. You cannot use the space mouse any more._

foreachthing, 2022-02-04:
Works for me.
2.4.1 b1, Win11
3DxWare 10.8.4.3407 and obsolete SapcePilot Pro

bubnikv, 2022-02-04:
> But when You tick any menu or dialog on PrusaSlicer it is over. You cannot use the space mouse any more.

What do you mean with that? What happens exactly?

Please note also, that the current 3DConnexion driver ships with invalid configuration specific to PrusaSlicer, see

https://github.com/prusa3d/PrusaSlicer/issues/7549#issuecomment-1006359089

sturmth, 2022-02-04:
Thank You. #7549 was helpfull. After editing the .xml-file the mouse continues to work.

closed 2026-09-29 by prusaslicer-bot[bot]